labgrid-project/labgrid #27: NetworkSerialPort sometimes sends characters in wrong order

tretter, 2017-03-15:
I'm using the SerialDriver backed by the NetworkSerialPort to connect to a target console. The test tries to login as 'root' without a password.

Most of the time the login is successful, but sometimes it runs into a timeout and the test fails. The debug output contains the following messages, indicating that the test tried to login as 'orot' and failing to supply a password.

```
DEBUG: Write bytes (b'root\n')
DEBUG: Reading 1 (min 1) bytes with 5 timeout
DEBUG: Read bytes (b'o') or timeout reached
DEBUG: Reading 16 (min 1) bytes with 4.89918851852417 timeout
DEBUG: Read bytes (b'rot\r\r\nPassword: ') or timeout reached
DEBUG: Reading 1 (min 1) bytes with 4.79821252822876 timeout
DEBUG: Read bytes (b'') or timeout reached
```

I would not expect that the driver reorders the characters that are sent via the serial console.

Emantor, 2017-03-15:
We directly hand over those bytes to the RFC2217 implementation of pyserial, which in turn sends those out through its own network socket. As far as I can tell there is no reordering done in their code path as well, maybe it's a ser2net bug?

jluebbe, 2017-03-15:
Could you check the TCP packets using wireshark? Are they really transferred in the wrong order?

tretter, 2017-03-16:
I checked the TCP packets for the write and the read.

The data in the packet from my machine is "root.", but the data of the packet to my machine is "orot...".

Emantor, 2017-03-16:
After further discussion this issue is unrelated to @labgrid-project, closing.

tretter, 2017-03-17:
The issue is a bug in agetty, which potentially reorders the characters on the login prompt (https://bugzilla.kernel.org/show_bug.cgi?id=194905).

However, labgrid should be able to handle this situation and as a workaround send the username more slowly.

jluebbe, 2017-04-11:
I've implemented a [workaround](/jluebbe/labgrid/tree/wip/txdelay), which introduces a delay before sending each byte. That seems to solve this problem.
Before merging that workaround, we'll test it internally some more to see how robust it is.
If it works well I'll probably add a way to override the txdelay for specific transmissions (such as sending the user name in the ShellDriver).

jluebbe, 2017-05-19:
The workaround has been merged (see the 'txdelay' parameter for the ConsoleProtocol driver in the docs: http://labgrid.readthedocs.io/en/latest/configuration.html#drivers).

The original bug was fixed in util-linux and is contained in v2.30-rc1.

closed 2017-05-19 by jluebbe