labgrid-project/labgrid #66: ExternalPowerDriver fail to call cmd

jremmet, 2017-03-30:
Setting cmd_on to a String holding the command and parameters fails.
from examples/shell/remotelab-1004.env
    cmd_on: "ptx_remotelab -1 1004"

    >>> subprocess.check_call("ptx_remotelab -1 1004")
      FileNotFoundError: [Errno 2] No such file or directory: 'ptx_remotelab -1 1004  
     >>> subprocess.check_call("ptx_remotelab")
       0

Passing a list, like 
    cmd_on: ["ptx_remotelab", "-1", "1004"]
fails due to cmd_on = attr.ib(validator=attr.validators.instance_of(str))

What is the preferred way to fix this? shell=true should be avoided, Forcing lists in the config isn't nice. Maybe cmd_on_param ?

jluebbe, 2017-03-30:
Thanks for trying labgrid! Note, that we're sill working on completing the basic documentation.

The ExternalPowerDriver instance from remotelab-1004.env is not actually used in the shell example, so it should probably be moved somewhere more useful.

For the check_call, we should use the shlex.split() function in the ExternalPowerDriver (similar to the ExternalConsoleDriver) from the standard library as suggested here:
https://docs.python.org/3/library/subprocess.html#subprocess.Popen
That way we can still have a simple string in the environment config, but also use check_call without shell=True.

Emantor, 2017-04-04:
Should be fixed by #68, closing.

closed 2017-04-04 by Emantor