labgrid-project/labgrid #67: doc/usage.rst: Stategy Fixtures Example

jremmet, 2017-03-30:
I have trouble to get the "Stategy Fixtures Example" working. Using the proposed conftest.py it got:

```
target = Target(name='main', env=Environment(config_file='local.yaml'))

    @pytest.fixture(scope='function')
    def active_command(target):
>       return target.get_active_driver(CommandProtocol)

conftest.py:31: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = Target(name='main', env=Environment(config_file='local.yaml'))
cls = <class 'labgrid.protocol.commandprotocol.CommandProtocol'>

    def get_active_driver(self, cls):
        """
            Helper function to get the active driver of the target.
            Returns the active driver found, otherwise None.
    
            Arguments:
            cls -- driver-class to return as a resource
            """
        for drv in self.drivers:
            if isinstance(drv, cls):
                if drv.state == BindingState.active:
                    return drv
        raise NoDriverFoundError(
>           "no driver matching {} found in target {}".format(cls, self)
        )
E       labgrid.exceptions.NoDriverFoundError: no driver matching <class 'labgrid.protocol.commandprotocol.CommandProtocol'> found in target Target(name='main', env=Environment(config_file='local.yaml'))

/home/jremmet/labgrid/labgrid-venv/lib/python3.5/site-packages/labgrid-0.1.dev475+n450511c-py3.5.egg/labgrid/target.py:127: NoDriverFoundError
```

I also noted the in "examples/strategy/test_barebox_strategy.py"  CommandProtocol is replaced:

```
    #command = target.get_driver(CommandProtocol)
    command = target.get_driver(BareboxDriver
```

jluebbe, 2017-03-30:
The example in the documentation as currently written needs some code changes which require some more testing before I can merge them. I'll notify you in this issue as soon as that example should work with the code in master.

jremmet, 2017-03-30:
Thank you very much. labgrid looks fine so far.

jluebbe, 2017-04-11:
The example should be fixed since merging PR #80. @jremmet, could you check if it works for you now?

jremmet, 2017-04-12:
Works fine now.

closed 2017-04-12 by jremmet