labgrid-project/labgrid #106: Labgrid driver/resource binding does not work with new attrs behavior

jnejedly, 2017-06-29:
Labgrid is not compatible with the most recent version of attrs due to a change to the default hashing behavior (unhashable). See here: https://github.com/python-attrs/attrs/issues/136

The BindingMixin class defines suppliers and clients as sets, which cannot contain unhashable objects. I think the fix would be to either change all drivers/resources to use the @attr.s(cmp=False) option to disable generation of comparison methods and make the objects hashable, or change the way that the BindingMixin registers suppliers and clients.

Emantor, 2017-07-02:
This should be fixed on master with 43095183abf0ff7041170a86a54679c2b241c8f1, are you using the current release or master?
We are aware we currently depend on no longer supported behaviour, but have not discussed or looked into the issue yet.

jnejedly, 2017-07-05:
I was using v0.1.0, but when I was looking into this issue I saw that the master branch had the older attrs version specified. I just wanted to make sure you were aware which change in attrs caused the issue, so it could be fixed in a future release.

Emantor, 2017-07-26:
I posted a WIP branch #113, I would highly appreciate it if you could test labgrid with this branch applied. We are rolling the branch out for internal testing as well.

Emantor, 2017-08-01:
I accidently deleted the branch on github, new PR is #116.

closed 2017-08-01 by jluebbe

jnejedly, 2017-08-01:
Sorry, I meant to test this out. Looks like you were already able to internally test the changes, but I did just test them myself and it seems like it fixes the issue. Thanks!

commit 95168718 (2017-08-01): multiple: explicitly set attr.s cmp parameter

We mostly don't need the compare functionality of attrs. Disable it to fall back
to object id hashes (similar to python standard classes). The only exception is
the ResourceMatch class, since we collect these into a list and need to remove
them later on. The list remove function requires the implementation of the
compare functions, explicitly enable it for the ResourceMatch class.

Fixes #106