labgrid-project/labgrid #109: Labgrid for parallel target testing

jnejedly, 2017-07-10:
I've been evaluating labgrid for use in a new parallel hardware testing system. Unfortunately there are two main features that don't seem to be currently supported that would be required for our project:

1. **Multiple driver/resource instances on a single target**
This is listed as planned feature in the docs so I assume it's already being looked into. The use case I've been looking at is using a serial port for communication with the UUT and a second serial port for communication with a separate power control board. It might be possible to work around this by creating new resource/driver classes for the second device, but this does not seem like a good solution.
2. **Shared driver instances between multiple targets**
The power control board I mentioned above has multiple relays to control multiple target boards. I do not think it is currently possible to share a single serial device between boards like this, since bound resources/drivers cannot be controlled by another target.

Are these features planned to be supported?

jluebbe, 2017-07-11:
While we're looking into multiple instances of the same protocol per target, I suspect that it's not what you want for this use case: Each (per-device) output of the power control board should be represented at the target level as an implementation of the PowerProtocol, so that the higher levels don't need to know what sort of power switch a specific target uses.

We could hide the shared serial connection in the driver, similar to how the USBResources work with the UdevManager (one manager for several resources). This would have the disadvantage that only one process could use the power control board at a time. I see two ways to avoid this:

1. Use an external lock file and open the serial port for a short time when switching power from a custom driver.
2. Use an external process to provide multiplexed access to the board.

I'd prefer variant 2, as it will make it easier to also allow access to the board via the RemotePlace/coordinator. The pdudaemon (https://github.com/Linaro/pdudaemon) used by LAVA looks like it would be easy to add support for a board attached via serial. Then we'd just need to have the equivalent of https://github.com/Linaro/pdudaemon/blob/master/pduclient as a PowerProtocol driver.

jnejedly, 2017-07-11:
It seems to me that using an external process and RemotePlace would be more complicated than is necessary since in this case the power board serial port will always be on the local host machine. Is there any way to do this with only local resources/drivers? It seems like it wouldn't work since the CommandProtocol/ConsoleProtocol drivers for the shared serial connection would still only be able to be bound to a single target.

Sorry if I'm not understanding, I'm still very new to using labgrid.