labgrid-project/labgrid #137: SSH-connection to target without IP

jonau, 2017-09-21:
Is there a easy way to connect to the target via ssh without knowing its ip-address?
Currently we use:
```python
service = target.get_resource(NetworkService)
service.address = ip_address
ssh = target.get_driver(SSHDriver)
```
with a `NetworkService` and a `SSHDriver` defined in the environment.

I would like an approach where we have a new driver, which serves as a NetworkService and responds with the ip address gathered from the info-driver.

Emantor, 2017-09-22:
Pull request #83 adds a feature to retrieve the ip-address via SNMP from the connected network switch, this is however only feasible if you have SNMP access on the switch. And the PR is a days old and stale.

The InfoDriver already provides a way to retrieve the ip-address, could you rephrase your approach?

jonau, 2017-09-22:
The Problem arises when you try to connect to the target via ssh. As I keep my environment in the *.yam file I need to set the ip-address (or hostname) for the ssh connection in the `NetworkService`. At the moment I have a workaround by setting the ip address of the NetworkService again before I activate the SSHDriver.
My approach is to provide an alternative to the NetworkService. This would be a driver which has a binding to the InfoDriver to get the current ip address of the target. It then provides the ip-address together with the preconfigured username to the SSHDriver.

So instead of
```
NetworkService <- SSHDriver
```
we would have a Resource/Driver chain of:
```
SerialPort <- SerialDriver <- ShellDriver <- InfoDriver <- [new]TargetNetworkServiceDriver <- SSHDriver
```

To recap: The Problem is not the retrieval of the IP-Adress (provided #138 is solved) but to provide it in a dynamic way to the SSHDriver

Emantor, 2017-09-22:
That can be solved by retrieving the IP-Address in the strategy after the device is booted into userspace. After the ShellDriver activation, activate the InfoDriver, retrieve the IP, update the resource and then activate the SSHDriver.

jonau, 2017-09-25:
Ok this seems like a good approach.
Thanks.

closed 2017-09-25 by jonau