labgrid-project/labgrid #138: Info Driver find multiple matching drivers

jonau, 2017-09-21:
If I provide an info driver in my environment it raises an the error:
```
labgrid.exceptions.NoDriverFoundError: multiple drivers matching <class 'labgrid.protocol.commandprotocol.CommandProtocol'> found in target Target(name='main', env=Environment(config_file='env.yaml'))
```
This is due to the fact, that I have an SSH-Driver, an Barebox-Driver and an Serial Connection to the Target defined in the environment.
I suppose #124 will solve the problem, as we than can select the right driver.

I also saw the TODO in the source of InfoDriver.
```python
# TODO: rework CommandProtocol binding to select correct underlying driver
# (No UBoot/BareboxDriver, SSH > Serial,…)
```
Do you have a specific plan for that?

Emantor, 2017-09-22:
A test of #124 would be greatly appreciated, although the documentation for the new selection is still missing.

Regarding the priorities:
Yes we have, the current idea is to implement priorities in the bindings via integers. A driver can explicitly assign a priority to a supplier and thus get the best matching result. There is no code available yet though.

Emantor, 2018-02-12:
We merged multiple driver support since then and priorities are coming up soon, i'll close this for now. Don't hesitate to reopen if something comes up on your side.

closed 2018-02-12 by Emantor