labgrid-project/labgrid #139: SSH-Connection breaks when rebooting

jonau, 2017-09-22:
When I reboot the target with the barebox-driver during an active ssh-session, i get the following error while trying to send another command over ssh:
```
packet_write_wait: Connection to 10.171.8.15 port 22: Broken pipe
mux_client_request_session: read from master failed: Broken pipe
Host key verification failed.
   INFO: Socket already closed
```
I guess there is no way to circumvent the broken pipe problem, since the power gets cut off. However I would like labgrid to detect such a failure and try to reconnect. Currently I don't see an easy way to implement the behavior without changing the way the ssh-driver works internally. What would be an appropriate implementation here?

My current workaround is to simple deactivate and reactive the ssh-driver in the (pytest-)fixture which provides the ssh-driver. This works but since the ssh teardown and setup takes around 250ms it slows down the tests immensely.

Emantor, 2017-09-22:
Deactivating the driver here is the intended solution. If you move the deactivation of the SSHDriver into the strategy, you only deactivate it during the reboot and can later reacivate it if the device is booted into userspace again.

Labgrid does not know whether your device is going to reboot into userspace or into the bootloader, an automatic reconnect feature is not feasible.

jonau, 2017-09-25:
Ok, seams to be the better way to do it in the strategy.
Thanks.

closed 2017-09-25 by jonau