labgrid-project/labgrid #141: pip install labgrid[coordinator] dependency problem

krevsbech, 2017-09-29:
In order to run crossbar and the coordinator I've ran:
`pip3 install labgrid[coordinator] `
Which works however it installs idna-2.6 which is not valid for crossbar, crossbar require =>2.5, <2.6 
Manually uninstalling idna-2.6 and installing idna-2.5 works.

`pip3 uninstall idna
pip3 install idna==2.5`
I will be glad to help solving the issue upstream however, that would require a bit help in order to pointing at the place where the requirement is described.

Emantor, 2017-10-06:
Could you try with current master? I could not reproduce it there. Thanks.

krevsbech, 2017-10-09:
Using latest master of labgrid and the pip released labgrid[coordinator] I get:

`pkg_resources.VersionConflict: (idna 2.6 (/opt/labgrid-venv/lib/python3.5/site-packages)Requirement.parse('idna<2.6,>=2.5'))
`
I don't know if it is a pip thing or if I am not supposed to use a released version of labgrid[coordinator] , Hence I've created a docker file that reproduce the problem and describe my steps:
```
FROM ubuntu:16.04
RUN apt-get update
RUN apt-get install --assume-yes git python3 virtualenv python3-pip libffi-dev python-dev

RUN virtualenv -p python3 /opt/labgrid-venv && . /opt/labgrid-venv/bin/activate && pip3 install labgrid[coordinator] xmodem

RUN git clone https://github.com/labgrid-project/labgrid /opt/labgrid

````
Then:
```
docker build -t labgrid-coordinator-reproduce .
docker run -it -a stdin -a stdout --name labgrid-coordinator-container labgrid-coordinator-reproduce /bin/bash
```
Inside the container:
```

source /opt/labgrid-venv/bin/activate
cd /opt/labgrid
crossbar start

````

Emantor, 2017-10-09:
Ah, I finally was able to reproduce the problem. Apparently the dependency for crossbar is not checked if it is used as `extra_requires`. `requests` already installs `idna` but in a version which is not compatible with `crossbar`. Since we install `requests` first, crossbar complains afterwards.

There is already an upstream bug: crossbario/crossbar#1196. I'll mark this as upstream.

Emantor, 2017-10-10:
Upstream receommends to install crossbar standalone into its own virtualenv. Our coordinator still requires part of the labgrid library, specifically `labgrid.remote.coordinator`, I'll think on that.

jluebbe, 2017-10-19:
Could we fix idna to the same version as crossbar for now?

Emantor, 2017-10-19:
The problem is that we as a library can depend on other libraries using ranges, but crossbar as an application pins their libraries to specific versions. I'm more inclined to move the coordinator installation out of the extra and into an extra setup call. I think we'll run into similar problems in the long run if we start pinning libraries versions we don't require in our setup file.

jluebbe, 2017-10-20:
We currently use a guest component in Crossbar and also provide an authenticator, so we can't easily switch to separate virtualenv.

I'd like to do the split after 0.2.0, if a workaround is possible for now.

PR #145 (2017-10-22): setup: workaround for the coordinator extra install
This is a workaround for Issue #141. We'll introduce a proper fix later on.

Fixes #141
closed 2017-10-22 by jluebbe