labgrid-project/labgrid #142: Dockers files

krevsbech, 2017-10-04:
**Proposal:**
I've created a docker file to setup a coordinator in a container and one to run a labgrid-client/execute the tests. 
Are you interested in these, and if yes. Should i create a PR adding the folder docker to the labgrid repository, or are you more up for a separate repository inside the project ?

Emantor, 2017-10-09:
We are definitely interested, I would prefer a separate repository in the project.

krevsbech, 2017-10-09:
Perfect 
I've pushed to my own space for now as I don't think i have the permissions to create a repository in your project. Will you create a repository that I can put a PR in, or clone: https://github.com/krevsbech/labgrid-docker-images

krevsbech, 2017-10-18:
@Emantor Will you crate a repo in the project space, then I will make a PR with the docker files?

Emantor, 2017-10-19:
There is https://github.com/labgrid-project/labgrid-docker-images now, please submit your PR there, sorry for the wait.

krevsbech, 2017-10-20:
No problem.
Can you please create an empty Master branch with e.g. a readme, as I am not allowed to open a pull request where there is no branch.

jluebbe, 2017-11-29:
I've deleted the empty repo and forked your labgrid-docker-images instead. You should now be able to open PRs.

closed 2017-11-29 by jluebbe

krevsbech, 2017-12-01:
@jluebbe  Perfect
If you are interested I would be happy to help asses incoming pull request, on the docker repo.