labgrid-project/labgrid #150: Resource on exporter not available deserves a meaningful error message

saschahauer, 2017-11-06:
Situation: When a resource is exported on an exporter, but is currently not available (due to USB cable to the USB serial converter not connected for example), a meaningful error message should be printed when trying to connect to this resource. Instead, the following backtrace is printed:

Traceback (most recent call last):
  File "/opt/venv/python3.5-x86_64/lib/python3.5/site-packages/labgrid/remote/client.py", line 893, in main
    args.func(session)
  File "/opt/venv/python3.5-x86_64/lib/python3.5/site-packages/labgrid/remote/client.py", line 596, in console
    res = self._console(place)
  File "/opt/venv/python3.5-x86_64/lib/python3.5/site-packages/labgrid/remote/client.py", line 571, in _console
    target = self._get_target(place)
  File "/opt/venv/python3.5-x86_64/lib/python3.5/site-packages/labgrid/remote/client.py", line 523, in _get_target
    RemotePlace(target, name=place.name)
  File "<attrs generated init a6e6d25803fd8b485c18a2103d7ba0a40859b244>", line 12, in __init__
    self.__attrs_post_init__()
  File "/opt/venv/python3.5-x86_64/lib/python3.5/site-packages/labgrid/resource/remote.py", line 89, in __attrs_post_init__
    super().__attrs_post_init__()
  File "/opt/venv/python3.5-x86_64/lib/python3.5/site-packages/labgrid/resource/common.py", line 89, in __attrs_post_init__
    self.manager._add_resource(self)
  File "/opt/venv/python3.5-x86_64/lib/python3.5/site-packages/labgrid/resource/common.py", line 65, in _add_resource
    self.on_resource_added(resource)
  File "/opt/venv/python3.5-x86_64/lib/python3.5/site-packages/labgrid/resource/remote.py", line 46, in on_resource_added
    remote_place.target, resource_entry.cls, resource_entry.args)
  File "/opt/venv/python3.5-x86_64/lib/python3.5/site-packages/labgrid/factory.py", line 22, in make_resource
    r = self.resources[resource](target, **args)
TypeError: __init__() missing 2 required positional arguments: 'host' and 'port'

jluebbe, 2017-11-17:
This isn't what should happen for unavailable resources. They should have a valid host and port should be None. It seems to work:
```
$ labgrid-client -vv r guave/rl3-usb-p8/NetworkSerialPort
Exporter 'guave':
  Group 'rl3-usb-p8' (guave/rl3-usb-p8/*):
    Resource 'USBSerialPort' (guave/rl3-usb-p8/NetworkSerialPort[/USBSerialPort]):
      {'acquired': None,
       'avail': False,
       'cls': 'NetworkSerialPort',
       'params': {'extra': {'path': None}, 'host': 'guave', 'port': None}}
```
I've added a more specific exception to this case, so we can find the underlying problem.

Emantor, 2019-07-05:
This now errors correctly with:
```
labgrid-client: error: Not all resources are available: {NetworkSerialPort(target=Target(name='hello', env=None), name='USBSerialPort', state=<BindingState.bound: 1>, avail=False, host='dibooki', port=52249, speed=115200, protocol='rfc2217')}
This may be caused by disconnected exporter or wrong match entries.
You can use the 'show' command to all matching resources.
```

closed 2019-07-05 by Emantor