labgrid-project/labgrid #151: regular termination of microcom should not lead to a backtrace

saschahauer, 2017-11-06:
When microcom is ended using keyboard shortcut (microcom ends successfully in this case) no backtrace should be printed as this is normal and expected usage. Instead we get this backtrace:

sha-phycore-imx6sha@dude:~ labgrid-client console
connecting to  NetworkSerialPort(target=Target(name='sha-phycore-imx6', env=Environment(config_file='/opt/bin/labgrid.yaml')), state=<BindingState.bound: 1>, avail=True, host='guave', port=37055, speed=115200)
connected to 2001:67c:670:100:feaa:14ff:fe6a:9213 (port 37055)
Escape character: Ctrl-\
Type the escape character followed by c to get to the menu or q to quit

Enter command. Try 'help' for a list of builtin commands
-> ^Cexiting
Traceback (most recent call last):
  File "/opt/venv/python3.5-x86_64/lib/python3.5/site-packages/labgrid/remote/client.py", line 893, in main
    args.func(session)
  File "/opt/venv/python3.5-x86_64/lib/python3.5/site-packages/labgrid/remote/client.py", line 596, in console
    res = self._console(place)
  File "/opt/venv/python3.5-x86_64/lib/python3.5/site-packages/labgrid/remote/client.py", line 587, in _console
    "{}:{}".format(resource.host, resource.port)
  File "/usr/lib/python3.5/subprocess.py", line 249, in call
    return p.wait(timeout=timeout)
  File "/usr/lib/python3.5/subprocess.py", line 1389, in wait
    (pid, sts) = self._try_wait(0)
  File "/usr/lib/python3.5/subprocess.py", line 1339, in _try_wait
    (pid, sts) = os.waitpid(self.pid, wait_flags)
KeyboardInterrupt

saschahauer, 2017-11-06:
It turned out I end microcom with Ctrl-\ Ctrl-c. In this case the Ctrl-c is no longer delivered to microcom (which would then restore the terminal settings), but to Python, which then prints the backtrace. microcom is then ended without the signal handler being executed and the terminal may be garbled.
I'll have to reprogram my fingers to end microcom with Ctrl-\ Ctrl-\ and this problem does not show up

jluebbe, 2017-11-06:
Would this work better if labgrid would send SIGTERM to the microcom process?

saschahauer, 2017-11-06:
Look at /usr/lib/python3.5/subprocess.py:249:

def call(*popenargs, timeout=None, **kwargs):
    """Run command with arguments.  Wait for command to complete or
    timeout, then return the returncode attribute.

    The arguments are the same as for the Popen constructor.  Example:

    retcode = call(["ls", "-l"])
    """
    with Popen(*popenargs, **kwargs) as p:
        try:
            return p.wait(timeout=timeout)
        except:
            p.kill()
            p.wait()
            raise

Here is the problem: Both python and microcom receive the Ctrl-C. If python comes to first handle the Ctrl-C then microcom will be killed and thus cannot cleanup the terminal

Emantor, 2018-11-06:
@saschahauer is this a still a problem that needs a workaround? Otherwise I'll close this issue.

Emantor, 2019-02-27:
Clsoing this as no longer requiring action

closed 2019-02-27 by Emantor