labgrid-project/labgrid #161: YKUSHPowerPort via coordinator fails

tgbloks, 2017-11-24:
H

i just tried to run my tests (using the new YKUSHPowerDriver) via the exporter / coordinator over network (the actual setup uses 3 different docker containers for coordinator, exporter and client where only the exporter has USB access) and it does not work - looks like the client is trying to access USB locally.
(YKUSHPowerDriver complains about not being able to access the USB port on the HW)
Should that work or did i miss to implement something?
here my configuration:
exporter:

foo:
  location: bar
  NetworkService:
    address: "10.10.10.10"
    username: "root"
  YKUSHPowerPort:
    serial: "serial"
    index: 1

client:

targets:
  main:
    resources:
      RemotePlace:
        name: "dev1"
    drivers:
      YKUSHPowerDriver: {}
      SSHDriver:
        keyfile: "./id_rsa"
      BloksStrategy: {}

jluebbe, 2017-11-24:
Yes, we don't have a generic way to control any exported devices.

For fastboot and imx-usb-loader, the client connects to the exporter host via SSH.
For serial ports, the exporter starts a ser2net process per port.
For OneWire, the exporter just refers to a ow-server running on the same host.
For NetworkPowerPorts, the exporter just provides the static information and the client connects to it directly.

I'm not sure how we could provide transparent access to any remote resource. I'd like to avoid implementing labgrid specific network protocols if at all possible.

For power control, using [pdudaemon](https://github.com/pdudaemon/pdudaemon) on the exporter and writing a PDUDaemonPowerPort driver seems to be the easiest way.

tgbloks, 2017-11-25:
Thx for the answer.
i will look into pdudaemon but it might just be easier for us to use different jenkins nodes on what was planned to become the exporters and run labgrid without the coordinator...

closed 2017-11-25 by tgbloks