labgrid-project/labgrid #162: Can't import NetworkSerialPort - doesn't exist

MartinSA18, 2017-11-27:
Hey,

I have an issue when trying to run or start anything regarding Labgrid, that i get an ImportError: cannot import name 'NetworkSerialPort'

Complete error message:
```
labgrid-venv/lib/python3.5/site-packages/labgrid-0.1.1.dev136+g57bc22d.d20171127-py3.5.egg/labgrid/resource/__init__.py", line 2, in <module>
    from .serialport import RawSerialPort, NetworkSerialPort
ImportError: cannot import name 'NetworkSerialPort'
```

Do i need to install "NetworkSerialPort" or something? When i go to the specified folder i don't see any files called "NetworkSerialPort" either - should there be one?

jluebbe, 2017-11-27:
``NetworkSerialPort`` is a class in [``labgrid/resource/serialport.py``](https://github.com/labgrid-project/labgrid/blob/master/labgrid/resource/serialport.py), so this should work. Especially if it can import RawSerialPort from the same file. How did you install labgrid?

MartinSA18, 2017-11-27:
I installed it by cloning the github repository using a virtual environment as per recommended in the installation instructions. Should i try to do it differently?

MartinSA18, 2017-11-27:
Also when i try to use the SerialDriver it gives me the following error:

`>>> sd = SerialDriver(t, name=None)
no resource matching <class 'labgrid.resource.serialport.NetworkSerialPort'> found in target Target(name='example', env=None)`

jluebbe, 2017-11-27:
> I installed it by cloning the github repository using a virtual environment as per recommended in the installation instructions. Should i try to do it differently?

Installing from github is fine. I couldn't reproduce the ImportError with the commit you are using (57bc22d9ac6df95aede7a4a3abf5c4f761684917), though.

> Also when i try to use the SerialDriver it gives me the following error:

How did you fix the original problem?

You found a bug: The message shouldn't appear, as adding the SerialDriver was successful:
```
>>> from labgrid.driver import SerialDriver
>>> sd = SerialDriver(t, name=None)
no resource matching <class 'labgrid.resource.serialport.NetworkSerialPort'> found in target Target(name='example', env=None)
>>> sd
SerialDriver(target=Target(name='example', env=None), name=None, state=<BindingState.bound: 1>, txdelay=0.0)
>>> 
```

MartinSA18, 2017-11-27:
I didn't really fix it. I just tried to install via Pip and not into a virtual environment. This helped in the sense that i can now run the coordinator (crossbar).  If i however try creating an exporter from a simple .yaml file i get the following error:

```
WARNING: session closed with reason wamp.error.no_auth_method [cannot authenticate using any of the offered authmethods ['ticket']]
2017-11-27T16:36:50 session closed with reason wamp.error.no_auth_method [cannot authenticate using any of the offered authmethods ['ticket']]
connection lost
```

Same thing if i try to display client resources:
```
WARNING: session closed with reason wamp.error.no_auth_method [cannot authenticate using any of the offered authmethods ['ticket']]
^CTraceback (most recent call last):
  File "/usr/local/bin/labgrid-client", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.5/dist-packages/labgrid-0.1.1.dev155+gff7da59-py3.5.egg/labgrid/remote/client.py", line 946, in main
    session = start_session(args.crossbar, "realm1", extra)
  File "/usr/local/lib/python3.5/dist-packages/labgrid-0.1.1.dev155+gff7da59-py3.5.egg/labgrid/remote/client.py", line 743, in start_session
    loop.run_until_complete(ready.wait())
  File "/usr/lib/python3.5/asyncio/base_events.py", line 375, in run_until_complete
    self.run_forever()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 345, in run_forever
    self._run_once()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 1276, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.5/selectors.py", line 441, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)
KeyboardInterrupt
```

So the message appears even though the creation of the SerialDriver works? So i should be able to move forward with "sd" as above?

jluebbe, 2017-11-27:
> WARNING: session closed with reason wamp.error.no_auth_method [cannot authenticate using any of the offered authmethods ['ticket']]

Did the coordinator and authenticator components start up correctly via `crossbar start`? You should see a line like this:
`started component: labgrid.remote.authenticator.AuthenticatorSession id=6432222431396741`
No exceptions should be thrown when starting.

> So the message appears even though the creation of the SerialDriver works? So i should be able to move forward with "sd" as above?

Yes, the message only appears because SerialDriver supports two kinds of serial resources and the first one is not found, while the second one is.

MartinSA18, 2017-11-27:
No it doesn't. It seems the crossbar installation is isolated from the labgrid installation. i have the .crossbar folder in the labgrid repository but when i start crossbar nothing implies it to be connected to labgrid. How can i link them?

jluebbe, 2017-11-29:
The configuration file in the .crossbar folder should cause it to load labgrid.remote.authenticator.AuthenticatorSession as a component and the labgrid.remote.coordinator module as a "guest" process. That should show up in the crossbar log output.

MartinSA18, 2017-12-01:
I needed to run the command "crossbar init". That linked the configuration file to the crossbar installation and that now starts up correctly. I can't however use any of the drivers to connect to my board, since it doesn't have a login prompt or username, so i am going to write my own.

jluebbe, 2017-12-01:
OK, so your board just gives you a shell after booting? The ShellDriver could be changed to also look for the shell prompt pattern in parallel to the login prompt.

MartinSA18, 2017-12-04:
But what exactly is meant by the login prompt? Because there is no login. It boots straight up into the prompt "XXX # ".

jluebbe, 2017-12-04:
In https://github.com/labgrid-project/labgrid/blob/master/labgrid/driver/shelldriver.py#L108, the shell driver expects either the login prompt or the shell prompt. So you can use something like this in your config:
```
...
ShellDriver:
  prompt: 'XXX # '
  login_prompt: ' login: '
  username: 'root'
...
```
The login_prompt pattern will never match, but the (shell) prompt should. I'm not sure if we have tested boards without login recently, so that might be broken... Could you push your current code somewhere? You can also join the IRC channel #labgrid on freenode.

MartinSA18, 2017-12-04:
If i do that and try to run:

` pytest --lg-env local.yaml test_shell.py`

i get the following error:

`labgrid.exceptions.NoDriverFoundError: no driver matching <class 'labgrid.protocol.consoleprotocol.ConsoleProtocol'> found in target Target(name='main', env=Environment(config_file='local.yaml'))
`

I have created a repository on github and pushed som example code to. Does anything look weird here:
https://github.com/MartinSA18/labgrid_test.git

Emantor, 2017-12-15:
AFAIR we solved your problems on IRC, I'll close this Issue for now. If you encounter the bug again don't hesitate to reopen 👍

closed 2017-12-15 by Emantor