labgrid-project/labgrid #166: Stable release

kjeldflarup, 2017-12-04:
Is there a stable release of labgrid?

I tried to follow https://labgrid.readthedocs.io/en/latest/getting_started.html but ran into a version related error, and once that was fixed, I soon ran into another. 
Then I also tried the docker image, but that also suffered from upgrade issues.

I'm trying to start using labgrid, and don't really want to be on the bleeding edge. 
Thus I would like to know if there are some version recommendations regarding to labgrid, but certainly also regarding to which packages besides labgrid has been installed on my system.

jluebbe, 2017-12-04:
Yes, we had some issues with out dependencies (pytest, attrs and crossbar), and I'm not sure how to make sure that doesn't happen again. Probably we'll pin all dependencies for now. Then we can make a release that keeps working even if e.g. attrs or pytest change. If you join #labgrid on freenode, we'll try to help you get started.

One way or the other, I'll do a release soon. Just need to find some time to wrangle pypi again...

leopck, 2018-02-06:
Any update on this? Would love to have a stable release to deploy labgrid in our side as well. It would help having a fixed version and fixed dependencies that is known to work so that we could deploy in multiple sites for us.

krevsbech, 2018-05-01:
Would it be possible/beneficial to do a minor release tag  eg. v0.1.1 from current master as the dependencies seems to be pined in requirements.txt.
This would allow the use of a released and reproducible environment ?

Bastian-Krause, 2019-01-15:
@jluebbe released [v0.2.0](https://github.com/labgrid-project/labgrid/releases/tag/v0.2.0). I guess we will not do stable releases in the near future.

bgermann, 2019-05-13:
Will you release v0.2.0 on PyPI as well?

Emantor, 2019-07-05:
The primary problem we have with PyPI is that we can't test the dependencies installed by Pypi. This is why our official documentation recommends the installation into a virtualenv using the requirements files. I can see however that pushing to pypi would be beneficial.

Emantor, 2021-01-16:
I just struggled with the github interface to create a `v0.3.0-rc1`, I will release the final version on friday next week.

closed 2021-01-16 by Emantor