labgrid-project/labgrid #175: Error when starting labgrid-client: ValueError: No mandatory attributes allowed after an attribute....

leonidlezner, 2017-12-21:
After installing labgrid via pip, when I execute labgrid-client, I'm getting following error :

```
Traceback (most recent call last):
  File "/home/pi/.virtualenvs/labgrid/bin/labgrid-client", line 7, in <module>
    from labgrid.remote.client import main
  File "/home/pi/.virtualenvs/labgrid/lib/python3.5/site-packages/labgrid/__init__.py", line 1, in <module>
    from .target import Target
  File "/home/pi/.virtualenvs/labgrid/lib/python3.5/site-packages/labgrid/target.py", line 6, in <module>
    from .driver import Driver
  File "/home/pi/.virtualenvs/labgrid/lib/python3.5/site-packages/labgrid/driver/__init__.py", line 3, in <module>
    from .serialdriver import SerialDriver
  File "/home/pi/.virtualenvs/labgrid/lib/python3.5/site-packages/labgrid/driver/serialdriver.py", line 10, in <module>
    from ..resource import SerialPort, NetworkSerialPort
  File "/home/pi/.virtualenvs/labgrid/lib/python3.5/site-packages/labgrid/resource/__init__.py", line 7, in <module>
    from .udev import USBSerialPort
  File "/home/pi/.virtualenvs/labgrid/lib/python3.5/site-packages/labgrid/resource/udev.py", line 148, in <module>
    class USBSerialPort(SerialPort, USBResource):
  File "/home/pi/.virtualenvs/labgrid/lib/python3.5/site-packages/attr/_make.py", line 644, in attrs
    return wrap(maybe_cls)
  File "/home/pi/.virtualenvs/labgrid/lib/python3.5/site-packages/attr/_make.py", line 613, in wrap
    builder = _ClassBuilder(cls, these, slots, frozen, auto_attribs)
  File "/home/pi/.virtualenvs/labgrid/lib/python3.5/site-packages/attr/_make.py", line 345, in __init__
    attrs, super_attrs = _transform_attrs(cls, these, auto_attribs)
  File "/home/pi/.virtualenvs/labgrid/lib/python3.5/site-packages/attr/_make.py", line 311, in _transform_attrs
    .format(a=a)
ValueError: No mandatory attributes allowed after an attribute with a default value or factory.  Attribute in question: Attribute(name='target', default=NOTHING, validator=None, repr=True, cmp=True, hash=None, init=True, convert=None, metadata=mappingproxy({}), type=None)
```

OS: Raspbian, Mac OS
Python: Python 3.5, Python 3.6

Emantor, 2017-12-22:
An update to attrs breaks labgrid, see #165, we plan to do a release soon, but not before next year. For now please use the git master as described in the development docs.

Emantor, 2018-02-12:
If the dev installation did not fix the problem please reopen this issue. I'll close this for now since a stable release is already tracked in #166.

EugenWiens, 2018-02-12:
I have seen this bug to day with the 'master'
`
(labgrid-venv) wiens@es-lxwiens2:~/proj/BspTests/labgrid$ labgrid-client --help
Traceback (most recent call last):
  File "/home/wiens/proj/BspTests/labgrid-venv/bin/labgrid-client", line 6, in <module>
    from labgrid.remote.client import main
  File "/home/wiens/proj/BspTests/labgrid-venv/lib/python3.5/site-packages/labgrid/__init__.py", line 1, in <module>
    from .target import Target
  File "/home/wiens/proj/BspTests/labgrid-venv/lib/python3.5/site-packages/labgrid/target.py", line 6, in <module>
    from .driver import Driver
  File "/home/wiens/proj/BspTests/labgrid-venv/lib/python3.5/site-packages/labgrid/driver/__init__.py", line 3, in <module>
    from .serialdriver import SerialDriver
  File "/home/wiens/proj/BspTests/labgrid-venv/lib/python3.5/site-packages/labgrid/driver/serialdriver.py", line 10, in <module>
    from ..resource import SerialPort, NetworkSerialPort
  File "/home/wiens/proj/BspTests/labgrid-venv/lib/python3.5/site-packages/labgrid/resource/__init__.py", line 7, in <module>
    from .udev import USBSerialPort
  File "/home/wiens/proj/BspTests/labgrid-venv/lib/python3.5/site-packages/labgrid/resource/udev.py", line 148, in <module>
    class USBSerialPort(SerialPort, USBResource):
  File "/home/wiens/proj/BspTests/labgrid-venv/lib/python3.5/site-packages/attr/_make.py", line 702, in attrs
    return wrap(maybe_cls)
  File "/home/wiens/proj/BspTests/labgrid-venv/lib/python3.5/site-packages/attr/_make.py", line 671, in wrap
    builder = _ClassBuilder(cls, these, slots, frozen, auto_attribs)
  File "/home/wiens/proj/BspTests/labgrid-venv/lib/python3.5/site-packages/attr/_make.py", line 371, in __init__
    attrs, super_attrs = _transform_attrs(cls, these, auto_attribs)
  File "/home/wiens/proj/BspTests/labgrid-venv/lib/python3.5/site-packages/attr/_make.py", line 337, in _transform_attrs
    .format(a=a)
ValueError: No mandatory attributes allowed after an attribute with a default value or factory.  Attribute in question: Attribute(name='match', default=NOTHING, validator=<instance_of validator for type <class 'dict'>>, repr=True, cmp=True, hash=False, init=True, metadata=mappingproxy({}), type=None, converter=None)
`

Emantor, 2018-02-12:
Yes, the fix for `attrs 17.4.0` is included in PR #189 as ecbdf6c7e232e9050e3c63462fef2e0d9fe4483f . Unfortunately the attrs library had to include a breaking change in their release.