labgrid-project/labgrid #176: SerialDriver driver activates U-boot prompt.

kjeldflarup, 2017-12-31:
Sometimes the SerialDriver stops u-boot. It seems to press the anykey, and gets the u-boot prompt.
Looking in _await_login this comment "# TODO use step timeouts" indicates that this is a known problem.

The interruption of u-boot is caused by _await_login calling self.console.sendline("") as the first thing.  
It is of course target dependent, but my board always prints login:, no need to send a newline to get it. Should the target already have passed that point, then say 500 ms silence would indicate that I need to send a newline, to get the current prompt, whether it would be the logged in prompt, the login prompt OR a u-boot prompt.

Thus _await_login should loop on self.console.expect and only send a newline when receiving a timeout and the serial port has been quiet during that period.
I had this behavior implemented in the Ruby code which I used before starting to use labgrid.

Furthermore it should be able to detect the U-boot prompt, and recover from that too. But if the interruption can be prevented, this is not necessary.

jluebbe, 2018-01-15:
This behavior is in ShellDriver, not in SerialDriver. It's there to trigger printing the current prompt, so that it can continue without waiting.
To avoid activating the ShellDriver on a U-Boot prompt, the robust way is to use the UBootDriver and a strategy to explicitly capture the board in U-Boot and then trigger the kernel boot.

kjeldflarup, 2018-01-18:
Explicit capture of UBoot, would mean that the test of UBoot is not complete. We might oversee a misconfiguration in Uboot.

jluebbe, 2018-03-19:
This should be fixed via PR #193.

closed 2018-03-19 by jluebbe