labgrid-project/labgrid #182: --lg-log affects pytest's rootdir

kjeldflarup, 2018-01-25:
Using the --lg-log option affects pytest's rootdir
```
(labgrid-venv) kfa@kjeld-Latitude-5580:/userdata/labgridtools/remotetests/remotetest$ pytest  --capture=sys -rs --lg-log /tmp/labgrid-log --lg-env local.yaml 
================================================================================== test session starts ==================================================================================
platform linux -- Python 3.5.2, pytest-3.3.2, py-1.5.2, pluggy-0.6.0
rootdir: /, inifile:
```
The tests runs ok, but the pytest cache does not work
`  could not create cache path /.cache/v/cache/lastfailed
`
It seems like pytest sets rootdir to be the commom topdir of --lg-log and current working direktory
```
(labgrid-venv) kfa@kjeld-Latitude-5580:/userdata/labgridtools/remotetests/remotetest$ pytest  --capture=sys -rs --lg-log /userdata/labgridtools/remotetests/log --lg-env local.yaml 
================================================================================== test session starts ==================================================================================
platform linux -- Python 3.5.2, pytest-3.3.2, py-1.5.2, pluggy-0.6.0
rootdir: /userdata/labgridtools/remotetests, inifile:

```

Emantor, 2018-02-15:
According to pytest-dev/pytest#1435 (mentioned in the [pytest docs](https://docs.pytest.org/en/latest/customize.html#finding-the-rootdir)) this is indeed the case for paths with default arguments. I'll create a pull request that fixes the behaviour.

Emantor, 2018-02-15:
Thinking further about this this is an issue of pytest. You'll have to set an explicit rootdir via one of the methods mentioned in the [pytest docs](https://docs.pytest.org/en/latest/customize.html#finding-the-rootdir), we can't change much about the fact that they iterate over all the pytest arguments to find a common rootdir.

kjeldflarup, 2018-02-16:
It seems that this is the way pytest works, and pytest has its own thought about it. So lets just close this.

closed 2018-02-16 by kjeldflarup