labgrid-project/labgrid #186: Single port binding for enterprise environments

leopck, 2018-02-06:
It seems that labgrid-client requires the use of multiple port number access to perform a simple console over Serial.

e.g.
  1. In order to connect to the coordinator, we need to pass a WebSocket URL with a randomized? port number.
```
labgrid-client -x ws://10.202.145.10:20408/ws -p home con
```
  2. Once the console is connecting, microcom will attempt to connect to the NetworkSerialPort via a port number that is automatically passed to microcom

```
microcom -s 115200 -t sirius02:48847
```

This post a problem for enterprise companies as in enterprise companies we have very strict regulations on port numbers. Lots of the port numbers are blocked off and labgrid-client console will not work in the enterprise environment.

Probably a good solution would be create a single port binding (e.g. 80, 443) and perform a port multiplexing, each port can be identified by a UID or a randomized token.

Any plans for this? I could help out in this if I can find the time.

leopck, 2018-02-06:
This project looks promising as a starting point. [https://github.com/khuevu/http-tunnel](url)

It looks to me that it is able to tunnel all the port from the server and client thru port 80.

jluebbe, 2018-02-07:
The coordinator port can be configured in .crossbar/config.yaml. Then you need to use the matching URL with labgrid-client -x.
For accessing the resources, the next step should be to use SSH port forwarding via the exporters. They are directly connected to the resources (either locally via USB or via a local network). By connecting to the exporter via SSH, you already have the functionality to execute local programs (fastboot, imx-usb-loader, dd) and can dynamically add port forwarding (for serial, ssh to the target, ...). This way, you can also reuse the existing auth mechanisms of SSH.

leopck, 2018-02-22:
@jluebbe not sure by what you meant with 

> to use SSH port forwarding via the exporters.

Any examples you could provide? My team is looking forward to this. 👍

leopck, 2018-02-26:
@jluebbe Any responds for this? I really hope this issue can be fixed ASAP.

jluebbe, 2018-09-25:
@leopck There is now the PR #308 which adds support for something like this. Could you check if that also covers your use case?

Emantor, 2019-05-24:
PR #428 has been merged and SSH can be used as a single transport for connections. Closing this, if you have any questions, reopen or chat with us in the #labgrid channel on freenode.

closed 2019-05-24 by Emantor