labgrid-project/labgrid #190: dd

kjeldflarup, 2018-02-12:


Emantor, 2018-02-12:
Can you provide a simple example? We do exactly the same in the `_start_own_master` function in the SSHDriver which establishes the connection and uses this Socket throughout the communication with the board.

kjeldflarup, 2018-02-13:
Maybe the controlpath is the problem, ill look into how that works. 
I do a reboot during the test. I like to do reboots in my tests.

Emantor, 2018-02-13:
Reboots will only work if you disable the SSHDriver after the reboot, are you using a strategy for rebooting?

kjeldflarup, 2018-02-13:
I got a fixture
```
@pytest.fixture(scope="function")
def resetshells(target):
    shell = target.get_driver('ShellDriver')
    shell.on_deactivate()
    shell.on_activate()
    shell = target.get_driver('SSHDriver')
    shell.on_deactivate()
    shell.on_activate()
```
I had commented out the SSH driver reset, because It apparently didn't have any effect, and I wasn't aware of the usage of multiplexing. When my own user knows about the fingerprint of the target, then the run command still works if the control path is lost.

closed 2018-02-13 by kjeldflarup