labgrid-project/labgrid #198: uboot strategy and multiple tests in pytest

GOMSPACEMartin, 2018-02-21:
Say I want to execute multiple pytests within the same test suite but I will have to bring the target into the  uboot console as first step in every test, how do I go about with this?

I tried with this simple example where I would need the target to be powered off/on and interrupted in the bootsequence where, before the test is executed, I will have an uboot prompt.

Assuming 'in_bootloader' would handle this i could just execute the whole suite : `pytest --lg-env fixture.yaml test_uboot_all.py` all in once.

Currently this will fail, while running the individual tests one by one will pass both tests:
`pytest --lg-env fixture.yaml test_uboot_all.py::test_uboot_1 
&& pytest --lg-env fixture.yaml test_uboot_all.py::test_uboot_2`

How can I achive the stragtegy putting my environment to 'default' for each test and not just pickup where the other test left off ?

**test_uboot_all.py :**
```
import pytest
import logging

from labgrid.protocol import CommandProtocol
from labgrid.strategy import UBootStrategy

@pytest.fixture()
def strategy(target):
    try:
        return target.get_driver(UBootStrategy)
    except:
        pytest.skip("strategy not found")

@pytest.fixture(scope="function")
def in_bootloader(strategy, capsys):
    with capsys.disabled():
        strategy.transition("uboot")

def test_uboot_1(target, in_bootloader):
    command = target.get_driver(UBootDriver)

    stdout, stderr, returncode = command.run('run dfu_sf')
    assert returncode == 0
    assert len(stdout) > 0
    assert len(stderr) == 0
    assert 'n25q512' in '\n'.join(stdout)

def test_uboot_2(target, in_bootloader):
    command = target.get_driver(UBootDriver)

    stdout, stderr, returncode = command.run('run dfu_mmc')
    assert returncode == 0
    assert len(stdout) > 0
    assert len(stderr) == 0
    assert 'found' in '\n'.join(stdout)
```

Emantor, 2018-02-26:
I interpret your question as "how can i get my device to reboot after the this test".

You would have to add another state to your strategy which executes the reboot and goes back into the bootloader. Writing your own strategy for your individual test case is necessary since we only provide very basic strategies to serve as examples.

jluebbe, 2018-03-19:
Does this answer your question, @GOMSPACEMartin?

GOMSPACEMartin, 2018-03-20:
@jluebbe only partly, I know that I will have to make advanced strategies for my tests, but when I include in_bootloader as part of inheritance I would still expect a strategy to make sure that I am in that state. as @Emantor wrote it is only examples provided, but I think that it must be a best practice example then.

I don't belive that it will be best practice to end/start in different states depending on the outcome of the last test.

@Emantor I am still trying to grasp the mechanisms to achieve the "how can i get my device to reboot after the this test" so that it is done in the intended way, e.g. how do I find the state that I ended up in after a transition?

jluebbe, 2018-03-26:
@GOMSPACEMartin Do you need to reboot your device after the test because it's in an unclean state? Or is testing the reboot part of your test case?

GOMSPACEMartin, 2018-03-26:
@jluebbe I do not need to reboot the device after the test, but I need to reboot the device before the test to bring it in a known state. 
I belive it is good practice to know your starting state, the way it is demonstrated, that state is not known before the test starts as the starting conditon will rely on the outcome of the last test executed.

Emantor, 2019-07-05:
I am closing this issue for now, if you have further questions don't hesitate to reopen, ask us on IRC/Matrix or send an e-mail to r.czerwinski@pengutronix.de.

closed 2019-07-05 by Emantor