labgrid-project/labgrid #203: Character decoding

kjeldflarup, 2018-02-27:
SOMEBODY thought that it was a good idea to use a unicode centerdot 0xb7 in a product name.

Now this gives me a failure in the SSHDriver: 
```
>       stdout = stdout.decode("utf-8").split('\n')
E       UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb7 in position 90: invalid start byte
```
The ShellDriver may have the same problem, as well as barebox and uboot  
Changing to iso-8859-1 solves my problem, for now.

Possible solutions.

1. Add an encoding parameter to the drivers.
2. Add an encoding parameter to the ressource
3. Add an encoding parameter to the run command
4. Change the error handling in the decode command to ignore or replace

The most optimal solution to the problem is to be able to set the encoding for the individual target. The easiest place may be the SSHDriver directly. But the info may actually be more accurate to set in the resource in the exporter, thus the same test case can connect to two differently configured targets.

A more simple solution would be to write ("utf-8", 'ignore')  
This though has the drawback, that stray binary output no longer causes the test to fail. This is clearly a tradeoff.  
The run command could be added an ignore parameter. (OR codecignore to be less confusing), in this way the failing test cases could be modified only.

jluebbe, 2018-03-19:
I've merged your PR #214, so you can configure the handling in the run() method. We can still allow specifying the default encoding on the driver later (which I'd prefer to the resource as it's not related to the physical interface).
Additionally, we could add add raw_run() methods (if needed) which wouldn't do any en-/decoding.
Also, the middle dot would need to be encoded as 0xC2 0xB7 (according to http://www.fileformat.info/info/unicode/char/b7/index.htm), so possibly the product name is buggy. ;)

closed 2018-03-19 by jluebbe

kjeldflarup, 2018-04-12:
Came across this issue else where  
The thing is that UTF-8 is a bad default iso-8859-1 (aka latin_1) is much better as default, because it is 8 bit, and all 256 characters would be legal, so no error could ever happen. Only use UTF-8 if special characters matters.