labgrid-project/labgrid #205: [question] Does labgrid support the concept of test repositories?

fkromer, 2018-02-28:
Does labgrid support the concept of test repositories to encapsulate test definitions from the test framework? E.g. the framework [avocado-framework/avocado](https://github.com/avocado-framework/avocado) includes builtin tests but provides the majority of tests in a separate test repository [avocado-framework/avocado-misc-tests](https://github.com/avocado-framework/avocado-misc-tests).

Emantor, 2018-02-28:
In so far that pytest supports them. Labgrid is meant as an interface for test libraries to access embedded Linux hardware. We don't ship tests aside from the examples. Does that answer your question?

fkromer, 2018-02-28:
Yes. First it was not obvious to me how to package custom tests. Tests, like the ones in the examples of labgrid usage [MartinSA18/labgrid_test](https://github.com/MartinSA18/labgrid_test) and [prevas-hnje/labgrid-example](https://github.com/prevas-hnje/labgrid-example) depend on labgrid via their pytest fixtures defined in their `conftest.py` file. `labgrid` just has to be defined as dependency of the custom test package as usual then...

Emantor, 2018-02-28:
I'll close this then :+1:

closed 2018-02-28 by Emantor