labgrid-project/labgrid #207: USBSerialPort should export IP address

kjeldflarup, 2018-03-01:
When using USBSerialPort gethostname is used to give the value of the "host" in the exported resource.
This assumes that the hostname of the system is defined in DNS.
OR the hostname should be set to an IP address.

A way to solve this would be to add a "host" parameter to USBSerialPort
```
USBSerialPort:
    match:
       'ID_SERIAL_SHORT': 'P-00-00682'
    speed: 115200
    host: 172.16.2.2
```
However the arguments for USBSerialPort seems to be divided into two objects.  
class USBResource in labgrid/resource/udev.py
class SerialPort in labgrid/resource/base.py

It looks to be wrong to add a host parameter to both.

Another solution would be to look for LG_HOSTNAME in the environment before calling gethostname()

Emantor, 2018-03-05:
Currently the host parameter is added by the exporter, see [exporter.py](https://github.com/labgrid-project/labgrid/blob/fcf88f4/labgrid/remote/exporter.py#98).

kjeldflarup, 2018-03-05:
@Emantor I noticed that, and I do this workaround currently:
`'host': os.environ.get("LG_HOSTNAME", gethostname()),
`

jluebbe, 2018-03-19:
We solved this differently in PR #218, by adding a `--hostname` argument to the exporter, which can be used to override the default system hostname.

closed 2018-03-19 by jluebbe