labgrid-project/labgrid #220: labgrid prints debug output in pytest without activation

SmithChart, 2018-03-18:
I am running pytests using a custom smallubootstrategy inside my project.

When I run pytest using
`pytest  --lg-env local.yaml test_uboot_strategy.py`
I get a lot of debug output from the Serial Output:
`DEBUG: Read 12 bytes: b'0x80000000\r\n', timeout 11.17, requested size 1`
`  DEBUG: Read 11 bytes: b'Loading: *\x08', timeout 11.17, requested size 1`

Since I have neither touched `logging.Logger.setLevel()` nor have used a `-v` on my command line I would just expect some pytest output.

I am currently using latest master + my PullRequests that have not been meged by now.

Is this intended behavior? Or do you have any idea how to track that bug down?

jluebbe, 2018-03-19:
No, that's not intended and I don't see any obvious recent change which would have caused this. Could you try a bisect?

Emantor, 2018-03-29:
Hm, I could not reproduce this behaviour on a quick test with a local Riotboard and current master.
Can you post your test files?

Emantor, 2019-03-22:
Is this still an issue?

Emantor, 2019-05-24:
Closing this, could not reproduce.

closed 2019-05-24 by Emantor