labgrid-project/labgrid #225: Communication with CI-system

kkreitmair, 2018-03-29:
We are evaluating to include labgrid to our CI-system. The goal is, to have a full automated CI-system.
Is there a way that labgrid receives a notification to start setting up boards and start the testsuit for the specific target, after the buildsystem (buildbot) has finished the build?

Emantor, 2018-03-29:
Labgrid, more specifically pytest, is supposed to be run from inside the CI System.
The persistent service labgrid provides is only used for the management of boards, resources and clients, not for test scheduling.

What would be useful in this context would be a reservation system to schedule an upcoming test, mentioned under the ideas section in the documentation [here](http://labgrid.readthedocs.io/en/latest/development.html#remote-target-reservation).

Emantor, 2018-04-04:
@kkreitmair did this answer your question? If so please close this issue.

kkreitmair, 2018-04-04:
Sorry for my late response and thank you for your answer.

I was expecting a possible scenario like this:
Trough the labgrid-client, i can call the pytest-plugin to run a specific test on specific places.

With this feature, it doesn't matter if a physical user or a program is interacting with the resources.
Also it is possible to separate more the components build-system, test-system and the hardware control-system.

Is such a solution possible in the road map of labgrid project?

kkreitmair, 2018-04-04:
Thank you for that. Now the concept behind the design is more clear to me.

closed 2018-04-04 by kkreitmair