labgrid-project/labgrid #230: Missing authorized_keys causes "ShellDriver.run can not be called"

kjeldflarup, 2018-04-10:
If I remove .ssh/authorized_keys on my target I get an error:

This is the full call path of the error
```
target = Target(name='main', env=Environment(config_file='labgrid.yaml'))

    @pytest.fixture(scope='function')
    def strategy(target):
>       shell = target.get_driver('ShellDriver')

conftest.py:6: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
../../labgrid/labgrid/target.py:191: in get_driver
    self.activate(found[0])
../../labgrid/labgrid/target.py:433: in activate
    client.on_activate()
../../labgrid/labgrid/driver/shelldriver.py:69: in on_activate
    self._put_ssh_key(self.keyfile)
../../labgrid/labgrid/step.py:209: in wrapper
    _result = func(*_args, **_kwargs)
../../labgrid/labgrid/driver/shelldriver.py:252: in _put_ssh_key
    self._run_check("chmod 700 ~/.ssh/")
../../labgrid/labgrid/driver/commandmixin.py:37: in _run_check
    stdout, stderr, exitcode = self.run(cmd, timeout=timeout)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = ShellDriver(target=Target(name='main', env=Environment(config_file='labgrid.yaml')), name=None, state=<BindingState.bo...800', keyfile='/userdata/labgridtools/ssh/labgrid_rsa.pub', login_timeout=120, console_ready='', await_login_timeout=2)
_args = ('chmod 700 ~/.ssh/',), _kwargs = {'timeout': 30}

    @wraps(func)
    def wrapper(self, *_args, **_kwargs):
        if self.state is not BindingState.active:
            raise StateError(
                "{} can not be called ({} is in state {})".format(
>                   func.__qualname__, self, self.state.name)
            )
E           labgrid.binding.StateError: ShellDriver.run can not be called (ShellDriver(target=Target(name='main', env=Environment(config_file='labgrid.yaml')), name=None, state=<BindingState.bound: 1>, prompt='~ # ', login_prompt=' login: ', username='root', password='****', keyfile='/userdata/labgridtools/ssh/labgrid_rsa.pub', login_timeout=120, console_ready='', await_login_timeout=2) is in state bound)
```

The error can be prevented in two ways:

1) Change run to _run
https://github.com/labgrid-project/labgrid/blob/aeb049d99b585d336e34f768d636c87a908262f7/labgrid/driver/commandmixin.py#L37
`stdout, stderr, exitcode = self.run(cmd, timeout=timeout)`

2) Change the order, so state is changed before calling on_activate
https://github.com/labgrid-project/labgrid/blob/aeb049d99b585d336e34f768d636c87a908262f7/labgrid/target.py#L432
```
    client.on_activate()
    client.state = BindingState.active
```

Emantor, 2018-04-10:
This was introduced in 1deea1c71728f50644ffd7b7e709577f43e1d4c2 by me. The first solution is preferred here, since `_run_check` is a private function which does not require an active driver. 
However since the commandmixin is also used in drivers which don't implement a `_run` function yet we'll have to move those calls as well.
I'll fix this tomorrow, thanks for the report.

kjeldflarup, 2018-06-20:
@Emantor what is keeping You from merging Your fix?

Emantor, 2018-06-20:
I'm still missing a review from @jluebbe

esben, 2018-06-26:
@jluebbe It would be really nice to get this merged...

PR #237 (2018-07-03): multiple: add _run function, use _run in _run_check for the mixin
Commit 1deea1c71728f50644ffd7b7e709577f43e1d4c2 introduced a bug where the
internal _run_check function used the run function guarded by a check_active
decorator. Use _run for the run_check function, refactor _run functions for
drivers who did not have it and adjust the tests to mock the correct funtion.

Fixes #230
commit 1f89ed9f (2018-07-03): multiple: add _run function, use _run in _run_check for the mixin

Commit 1deea1c71728f50644ffd7b7e709577f43e1d4c2 introduced a bug where the
internal _run_check function used the run function guarded by a check_active
decorator. Use _run for the run_check function, refactor _run functions for
drivers who did not have it and adjust the tests to mock the correct funtion.

Fixes #230
closed 2018-07-03 by jluebbe