labgrid-project/labgrid #246: SSHDriver cannot execute run_check

kjeldflarup, 2018-05-24:
This code fails:
```
@pytest.fixture(scope='session')
## Returns a SSHDriver object to do ssh operations
def sshcommand(strategy):
    strategy.transition("ssh")
    return strategy.ssh

def test_run_check(sshcommand):
    sshcommand.run_check('echo ssh')
```

```
E               TypeError: got an unexpected keyword argument 'timeout'
/usr/lib/python3.5/inspect.py:2910: TypeError
```

The problem seems to be that CommandMixin calls run in this way:
```
stdout, stderr, exitcode = self.run(cmd, timeout=timeout)
```
But only ShellDriver implements timeout.

BastianStender, 2018-05-24:
1deea1c ("driver/multiple: move run_check function") introduced this bug.

BastianStender, 2018-05-25:
`UBootDriver` and `FakeCommandDriver` are missing the timeout keyword too.

closed 2018-06-08 by jluebbe