labgrid-project/labgrid #249: Featue: Selecting test cases based on available hardware/fixtures

leonidlezner, 2018-05-28:
To make the test environment really scalable, it would be great, if I could define for each test case, which dependencies it has. For example, when having a complete rack with switchable power supplies, IO boards and externel peripherals, I could run all tests. When running the tests on the desk, without having the full setup, I would define in a config, which HW I have and hereby filter the tests, that are executed (automatically).

Emantor, 2018-05-28:
Also in ideas [here](http://labgrid.readthedocs.io/en/latest/development.html#target-feature-flags)

ejoerns, 2018-06-06:
An Idea how I currently handle sharing test across multiple different boards using both pytest mark and target names.

In my confest.py:

``` python
def pytest_addoption(parser):
    parser.addoption('--target', action='store', dest='selectedtarget', default='board-a',
                      help='target to run tests for')

def pytest_runtest_setup(item):
    if isinstance(item, item.Function):
        targetmarker = item.get_marker('target')
        if targetmarker is not None:
            targetname = targetmarker.args[0]
            if targetname != item.config.getoption('selectedtarget'):
                pytest.skip("test only runs on target %s" %(targetname))
```

Then I can specify a test that should run on `board-a` only:

``` python
@pytest.mark.target('board-a')
def test_watchdog(in_shell, command):
  ...
```

With using the target's name, we can also perform parameterized tests:

``` python
test_mem = {"board-a": 8072072, "board-b": 771496, "board-c": 971496}

 def test_mem_size(target, in_shell):
     """Test if memory size is correctly recognized."""
     command = target.get_driver(ShellDriver)

     stdout, val, returncode = command.run("grep MemTotal /proc/meminfo | awk '{print $2}'")
     size = int(stdout[0])
     assert size == approx(test_mem[target.name], abs=20000)
```

Maybe that helps

PR #299 (2018-09-25): Feature Flags
- [x] Docs
- [x] Tests

Fixes #249
closed 2018-09-25 by jluebbe