labgrid-project/labgrid #260: Automatic regression test via Docker Hub

kjeldflarup, 2018-06-08:
A few times we have experienced that Labgrid got broken for our use. Thus we came up with this idea.

Setup a docker hub project like this one https://hub.docker.com/r/deifwpt/labgrid-client/~/settings/automated-builds/ which is linked to a repository.

This is our own, but it is only linked to our own repository where we put our Dockerfile. If the Docker file is moved to labgrid, then any changes in labgrid/master will cause a new docker build.

This docker build, can then be used to do nightly regression tests in our CI system.

We would not use the image for our "production" tests. We would setup a separate docker image and manually rebuild that when we need the latest labgrid.

This of course means that the dockerfiles made by @￼krevsbech would be moved to labgrid

krevsbech, 2018-06-13:
From my perspective this makes sense. I will be happy to integrate the current docker work into the labgrid repo. 
I have one concern/risk that we must be aware of, if the docker build becomes a part of the regression testing on PR's, then if there are problems in the dependency chain of PIP libs it might seem like it is the diff in the PR breaking something, but the real problem is actually an upstream change of the libs. However, as you have already put an effort into mitigating this issues (By fixing versions), I see this as an acceptable risk.

kjeldflarup, 2018-06-29:
We just implemented this for our test production.

On Dockerhub/cloud I created https://hub.docker.com/r/deifwpt/labgrid with autobuild reference to changes in https://github.com/deif-wpt/labgrid

https://github.com/deif-wpt/labgrid is currently a fork of https://github.com/labgrid-project/labgrid, but we will remove the fork. Whenever we have a PR against labgrid, we may force push this to https://github.com/deif-wpt/labgrid master, likewise we will take the latest from labgrid-project and push that too. Thus we do not have an automatic link to labgrid-project

`deifwpt/labgrid` is our primary docker image. It could also have been a docker image administered by @jluebbe and friends.  
The docker file https://github.com/deif-wpt/labgrid/blob/master/docker/client/Dockerfile is a minimal labgrid build. 

If more packages are needed, then we have our production docker image: https://hub.docker.com/r/deifwpt/labgrid-client/ with this Dockerfile
```
FROM deifwpt/labgrid:latest
RUN pip3 install filemagic
```
Here we may add APT and Python packages not related to labgrid.

Obviously we could also on docker image deifwpt/labgrid-client have a branch and tag labgridregression
```
FROM labgrid-project/labgrid:latest
RUN pip3 install filemagic
```
Beside our normal tests using `deifwpt/labgrid-client:latest`, we could then also nightly run a test with `deifwpt/labgrid-client:labgridregression`. If that regression fails and my own latest does not, then I would start looking at labgrid and not my own code.

PR #320 (2018-10-18): Build Docker images
This add Dockerfile for building Docker images including the Labgrid distributed infrastructure components, and integrates build testing of them in Travis CI.

Replaces #284
Closes #260
closed 2018-10-18 by jluebbe