labgrid-project/labgrid #266: Logo for labgrid

blinry, 2018-06-12:
I heard labgrid needs a pretty logo :) Here are some rough sketches around one idea. What do you think? Is this a good direction? Which one do you like best? The logos on the left are logos of other Pengutronix projects, so you can see how the new one would fit in.

[image]

Note that this is not the final font. Can you find out which font was used for the other logos?

rohieb, 2018-06-13:
The font in the logos is [Khand](https://fonts.google.com/specimen/Khand), but it seems the serifs for the uppercase i were customized.

blinry, 2018-06-18:
I got offline feedback that the Erlenmeyer flask doesn't seem like a good fit for a technical project. ;) Any other ideas which come to mind when thinking of a *digital* lab? Generally speaking, would you prefer a logo which resembles some concrete object, or maybe a purely abstract one?

GOMSPACEMartin, 2018-08-20:
@blinry a 'pcb layout style' icon combined with some kind of grid perhaps?
[image] 
[image]

kjeldflarup, 2018-08-21:
Four chips in a grid? The challenge is that logo's sometimes are scaled down.

blinry, 2019-03-19:
After working on this issue with Pengutronix folks today, we built and decided on this logo:

[image]

closed 2019-03-19 by blinry