labgrid-project/labgrid #267: Exclusive access to resources

maximilianriemensberger, 2018-06-13:
We are currently evaluating using labgrid in a distributed deployment with multiple racks consisting of either one or multiple boards.  The use case we are interested in is allowing multiple developers and a CI system to access the racks / boards / resources on boards in a way such that they do not interfere with each other.

I'm currently hitting the following problem: I got a rack with two boards and tests that either require one board or both boards.  How am I suposed to make sure that I can lock each board individually (e.g. as an individual place) but also the whole rack of boards at once (as a single place)?  Currently, if I create two places that match some common resources I can lock both places thereby granting access simultaneous access to the same board to multiple developers and the CI system.  That's what I need to avoid.  Is this a config issue or a labgrid issue? If it's a config issue, what is the best way to guarantee mutual exclusion and still be able to grab either the whole rack (as a place) or individual boards (as individual places) for tests?

Emantor, 2018-06-15:
Resources should be exclusive to one of the boards, shared resources are not supported, since we do not lock the resources, only the place in labgrid. What kind of resource is shared between the boards? Are you able to split up the resource in such a way that both boards can be accessed simultaneously?

For a rack scenario i would create a place for each rack device and if the whole rack needs to be locked just call the lock command multiple times.

maximilianriemensberger, 2018-06-22:
Thanks for the explanation.

The network is a shared resource in our setup as we plan on doing automated tests that involve multiple boards and the network between them.  The other use case we are looking at is large racks (up to 500 boards).   I had hoped that there is a way to hierarchically organize "places" in labgrid such that I can lock each board individually or groups of boards  or all boards at once, such that I do not lock any resource (any board place) more than once by properly defining places for each board, each group, and the full rack.  If I understand you correctly, there are no utilities for managing hierarchical resources / places in labgrid?

closed 2018-06-22 by maximilianriemensberger