labgrid-project/labgrid #277: [ssh] <defunct> after powercut

kjeldflarup, 2018-07-02:
The Raspberry Pi running the exporter had restarted for some reason, which will cut the power to the DUT. This left my test in the state below.
This situation seems to go unnoticed for 2590 seconds. 

```
root        68    67  0 08:05 pts/0    00:00:04 /usr/local/bin/python /usr/local/bin/pytest --junitxml=logs/junit.xml --capture=sys --log-format=%(asctime)s %(levelname)s %(message)s --log-date-format=%Y-
root        78     1  0 08:05 ?        00:00:00 [ssh] <defunct>
root        79     1  0 08:05 ?        00:00:00 [ssh] <defunct>
root        94     1  0 08:06 ?        00:00:00 [ssh] <defunct>
root        95     1  0 08:06 ?        00:00:00 [ssh] <defunct>
root       328     0  0 08:51 pts/1    00:00:00 bash
root       338    68  0 08:53 pts/0    00:00:00 ssh -x -o LogLevel=ERROR -i /opt/labgridtools/ssh/labgrid_rsa -o PasswordAuthentication=no -F /dev/null -o ControlPath=/tmp/labgrid-ssh-tmp-0tr9x1fg/control
```
After a looong time I see this on the console:
```
Result: ('10pcm52-bsp-tests/test_pcm52-bsp.py', 137, 'test_os'):	passed	8.158427000045776s
FE
Result: ('10pcm52-bsp-tests/test_pcm52-bsp.py', 145, 'test_rt'):	failed	2590.722511768341s
The authenticity of host '172.29.5.34 (172.29.5.34)' can't be established.
ECDSA key fingerprint is SHA256:1yH9KbZJg2W8agWsOcjprrlqM5Kcpq34Lbd0zHWEwtk.
```

jluebbe, 2018-09-25:
Do you know if this happened while the SSHDriver was in it's _run() method? That one doesn't have any timeout and we don't use ServerAliveInterval (but maybe we should).

kjeldflarup, 2018-09-25:
Unfortunately I have little information on it, as the test was executed from Gitlab CI.
I haven't seen the error since, so it is not a big problem, but there is something that may trigger again.

ServerAliveInterval sounds like a good idea

BastianStender, 2018-10-22:
I hope that this is gone with #310. If not, please reopen this issue.

closed 2018-10-22 by BastianStender