labgrid-project/labgrid #296: A `login_optional: false` property to force ShellDriver to expect login_prompt first

a3f, 2018-08-14:
I am writing a test that should probe for a successful login which looks like this:

```
Welcome to Buildroot
buildroot login: root
#
```

I added `login_prompt: 'login: '` and `prompt: '# '` to my YAML:
```
targets:
  main:
    resources:
      - RemotePlace:
          name: 'birne-rechts'
    drivers:
      - SerialDriver: {}
      - NetworkPowerDriver:
          delay: 5.0
      - ShellDriver:
          login_timeout: 30
          login_prompt: 'login: '
          prompt: '# '
          username: 'root'
      - loginStrategy: {}
```

But `# ` matches beforehand while in U-Boot:

```
## Flattened Device Tree blob at 21000000^M
```

Because, helpfully, the `ShellDriver` doesn't mind matching a shell `prompt` even when `login_prompt` has yet to to arrive:

https://github.com/labgrid-project/labgrid/blob/8af10923f3f44495d425904c28b90021be02eb9e/labgrid/driver/shelldriver.py#L108

https://github.com/labgrid-project/labgrid/blob/8af10923f3f44495d425904c28b90021be02eb9e/labgrid/driver/shelldriver.py#L120-L128

Specifying `prompt: '\n# '` is a work around, but making this configurable via a `login_optional:` property would be nice.

I also wonder if a `re.MULTILINE` regex where `^` and `$` match start and end of line respectively wouldn't be the better default.

(Cc: @BastianStender @strumtrar who helped with debugging)

Emantor, 2018-08-14:
Looking at at this pexpect [line](https://github.com/pexpect/pexpect/blob/c0a0003916bfea649fcc79f4567890719c1cb249/pexpect/spawnbase.py#L235), you should be able to pass in your own compiled regex functions, if i am not misreading it. However I doubt that multiline will work in this case because pexpect will still receive input byte by byte and won't be able to distinguish individual lines.

a3f, 2018-08-14:
@Emantor It wouldn't allow `$` to match end-of-stream, because as your noted there is none, but otherwise it should be a drop-in replacement for matching new lines, e.g. in the concrete example above: `prompt: '^# '`, no?

Would it be worthwhile to have labgrid pass a compiled multiline regex, so the meaning of `^` and `$` in the YAML regexes is more intuitive? Users can still use `\A` is they only want to match start-of-stream.

Emantor, 2018-08-14:
How did you construct your regex? It should be a bytes like object, pexpect encodes the searches to bytes before generating the regexes for the list.

BastianStender, 2018-08-15:
Yep, got it. I'd like to convert all occurrences of pexpect regexes (that are configurable via yaml configs) to compiled regexes with `re.MULTILINE` enabled where appropriate. I'll do this when I find some time for that.

Emantor, 2018-08-15:
Sure, sounds great, thank you :+1:

jluebbe, 2018-09-25:
> Specifying `prompt: '\n# '` is a work around, but making this configurable via a `login_optional:` property would be nice.
>
> I also wonder if a `re.MULTILINE` regex where `^` and `$` match start and end of line respectively wouldn't be the better default.

`'\n# ' is actually the correct way here, because we *want* to allow matching over multiple lines as well as ignoring stuff at the beginning of a line. It would be more robust to have a proper prompt configured on the target, though.
That the ShellDriver matches a prompt even before login_prompt is intentional, as it allows activating the driver even when the user is already logged in. This is useful when repeatedly running a single test without rebooting the system.

BastianStender, 2018-10-22:
I think there is no more discussion needed here. If I'm wrong please reopen this.

closed 2018-10-22 by BastianStender