labgrid-project/labgrid #324: SyntaxError: invalid syntax: ensure_future = asyncio.async

LeBlue, 2018-10-12:
Calling:
`labgrid-client --help`

fails with:
 ```
Traceback (most recent call last):
  File "labgrid-venv/bin/labgrid-client", line 11, in <module>
    load_entry_point('labgrid==0.1.dev1+g10c485c', 'console_scripts', 'labgrid-client')()
  File "labgrid-venv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 484, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "labgrid-venv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2714, in load_entry_point
    return ep.load()
  File "labgrid-venv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2332, in load
    return self.resolve()
  File "labgrid-venv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2338, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "labgrid-venv/lib/python3.7/site-packages/labgrid-0.1.dev1+g10c485c-py3.7.egg/labgrid/remote/client.py", line 18, in <module>
    from autobahn.asyncio.wamp import ApplicationSession
  File "labgrid-venv/lib/python3.7/site-packages/autobahn/asyncio/__init__.py", line 36, in <module>
    from autobahn.asyncio.websocket import \
  File "labgrid-venv/lib/python3.7/site-packages/autobahn/asyncio/websocket.py", line 52
    ensure_future = asyncio.async
                                ^
  SyntaxError: invalid syntax
  ```

This is a bug in 'autobahn' that was fixed in v18.8.1. Currently used version per requirements.txt file is v17.10.1.

Using the newer version fixes the issue.

See the fixing commit: [Python 3.7 made "async" a keyword which caused a SyntaxError](https://github.com/crossbario/autobahn-python/commit/9b6fb57e5c87a5e29cd880f752a30b9409d480c6)

Emantor, 2018-10-12:
We currently do not support python 3.7. Python 3.7 support is on my todo list, but requires more changes than updating the requirements.

LeBlue, 2018-10-12:
Ah ok.

closed 2018-10-12 by LeBlue