labgrid-project/labgrid #340: Error at teardown of pytest test session

jgoetzinger, 2018-10-30:
Since the introduction of the ssh manager and manageded files we get an error at the teardown of our pytest testsession every time we run it. 
When the labgrid pytestplugin calls sshmanager.close_all() the function: _run_socket_command("cancel") in util/ssh.py returns exit status 255 with the following error message:
Control socket connect(/tmp/labgrid-connection-y3kpr80j/control-172.18.198.1): No such file or directory

Let me know, if you need anything else.

BastianStender, 2018-10-30:
Hi, this sounds familiar. I created PR #336 to fix this. Please test it.

jgoetzinger, 2018-11-06:
Hi, I testet it and  PR #336 indeed solves the issue.

BastianStender, 2018-11-16:
#351 went mainline instead of #336. Can you please verify that this also solves your issue?

jgoetzinger, 2018-11-20:
Hi, yes #351 solves the problem.