labgrid-project/labgrid #350: Fix Managedfile upload location in the documentation

Emantor, 2018-11-08:
The directory is no longer `/tmp/labgrid`, it is `/tmp/labgrid-$USER`, the documentation should be fixed and contain an example tmpfiles cleanup entry and cron job.

ukleinek, 2018-11-08:
From an administrator's POV I'd prefer if the location was e.g. `/var/cache/labgrid` as `/tmp` is not supposed to hold big files. (Consider it being a small tmpfs that occupies RAM for little gain.) So maybe it's sensible to think about moving the cache location before adapting the documentation to the current situation.

Emantor, 2018-11-12:
The exporter should also check that the `/var/cache/labgrid` directory exists.

rohieb, 2018-11-12:
Could portability be a problem here? Can it be ensured that the exporter does not run a non-FHS tool like e.g. Windows?

closed 2019-03-04 by jluebbe