labgrid-project/labgrid #422: Target attributes for tests

Bastian-Krause, 2019-05-03:
I came across a use case with multiple boards sharing a test suite (or at least a part of a test suite). A lot of tests can be written in a generic way, but some require a parameter, e.g. expected memory size (defined per board).
To overcome this I added these "target attributes" as a key value store in the [options yaml section](https://labgrid.readthedocs.io/en/latest/man/device-config.html#options). But this is not ideal, because "options" are apply to the whole config (not target specific). Having multiple boards described in one config renders this hack impossible. Also Config.get_option() returns all values as strings, but I'd like to get the type as the yaml parser returns them (strings, lists, ..).

This is somewhat comparable to [feature flags](#299), but "target attributes" need a key value store, not just boolean flags.

jluebbe, 2019-05-03:
Yes, this is useful. It should also be possible to get these attributes (or tags) from the place configuration on the coordinator.

kjeldflarup, 2019-05-06:
I have many times thought of adding something like this by cloning the NetworkService. Difference would be that the attributes are dynamical. 
```
  NetworkService:
    address: '172.29.5.35'
```

Emantor, 2019-09-20:
Wil be fixed by #490

Emantor, 2020-12-03:
Was fixed by #490

closed 2020-12-03 by Emantor