labgrid-project/labgrid #425: IMXUSBDriver not loading SPL

chtavares592, 2019-05-10:
Hello, 

I'm trying to use labgrid to bootsrap my device. 

I'm using the following local.yaml file:

```
targets:
  main:
    resources:
      IMXUSBLoader:
        match:
          'sys_name': '1-3'
    drivers:
      ManualPowerDriver:
        name: '1-3'
      IMXUSBDriver:
        image: image-spl
tools:
  imx-usb-loader: imx_usb_loader/imx_usb
images:
  image-spl: SPL
```
and with this tests.py file :
```
def test_usb(target):
    r = target.get_resource('IMXUSBLoader')
    assert r.devnum is not None
    assert r.busnum is not None

def test_imx_load(target):
    bp = target.get_driver('BootstrapProtocol')
    bp.load()

```
Using this, all tests pass, however, the SPL image is not loaded on my device. How can I debug this issue?

Emantor, 2019-05-24:
Does imx-usb-loader with the file without Labgrid work correctly? Are you using the Barebox imx-usb-loader? I'm pretty sure that the different usb-loader tools can't be simply replaced, since they use different command line arguments.

otavio, 2019-05-24:
This was indeed the issue; we did a few minor local changes and got it to work with Boundary Devices' one. However, U-Boot uses two files for loading.

I'd discuss how to get this supported by labgrid. Should we add a new loader driver? or add an option to define the desired behavior?

Emantor, 2019-05-24:
Yes, I think this should be added as a new loader driver. I'd propose `UBootIMXUsbLoader`, the resources for the `IMXUSBLoader` can simply be reused. @jluebbe do you agree?

otavio, 2019-05-24:
I'd prefer to add a mode selection attribute to the existing one so we don't confuse the user with multiple drivers.

In fact, we can use the loader for different use cases, not limited to U-Boot so it'd be misleading.

Emantor, 2019-07-05:
Thats correct. Iam not completely sure that the functionality between both versions can be shared by the same driver.

Emantor, 2020-12-03:
We have support for the NXP UUU Loader now, does this driver solve this issue?

closed 2021-03-05 by Emantor