labgrid-project/labgrid #440: labgrid install fails because of pytest-runner

spark5, 2019-05-24:
Hi!

I have latest on labgrid. Sometimes labgrid fails to install, but sometimes works.
Can you please help me?

Here is the output when trying to install but it failed:

Collecting pip
  Using cached https://files.pythonhosted.org/packages/5c/e0/be401c003291b56efc55aeba6a80ab790d3d4cece2778288d65323009420/pip-19.1.1-py2.py3-none-any.whl
Installing collected packages: pip
  Found existing installation: pip 8.1.1
    Uninstalling pip-8.1.1:
      Successfully uninstalled pip-8.1.1
Successfully installed pip-19.1.1
Collecting attrs==17.4.0 (from -r labgrid/requirements.txt (line 1))
  Using cached https://files.pythonhosted.org/packages/b5/60/4e178c1e790fd60f1229a9b3cb2f8bc2f4cc6ff2c8838054c142c70b5adc/attrs-17.4.0-py2.py3-none-any.whl
Collecting jinja2==2.10 (from -r labgrid/requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/7f/ff/ae64bacdfc95f27a016a7bed8e8686763ba4d277a78ca76f32659220a731/Jinja2-2.10-py2.py3-none-any.whl
Collecting pexpect==4.6 (from -r labgrid/requirements.txt (line 3))
  Using cached https://files.pythonhosted.org/packages/89/e6/b5a1de8b0cc4e07ca1b305a4fcc3f9806025c1b651ea302646341222f88b/pexpect-4.6.0-py2.py3-none-any.whl
Collecting pyserial from https://github.com/labgrid-project/pyserial/archive/v3.4.0.1.zip#egg=pyserial (from -r labgrid/requirements.txt (line 4))
  Downloading https://github.com/labgrid-project/pyserial/archive/v3.4.0.1.zip (193kB)
Collecting pytest==4.0.2 (from -r labgrid/requirements.txt (line 5))
  Using cached https://files.pythonhosted.org/packages/19/80/1ac71d332302a89e8637456062186bf397abc5a5b663c1919b73f4d68b1b/pytest-4.0.2-py2.py3-none-any.whl
Collecting pyudev==0.21.0 (from -r labgrid/requirements.txt (line 6))
  Using cached https://files.pythonhosted.org/packages/bc/a2/31a07829acea8e70a28c247f43fa5d981229ae0f9edfeddedf52de00709b/pyudev-0.21.0.tar.gz
Collecting requests==2.19.1 (from -r labgrid/requirements.txt (line 7))
  Using cached https://files.pythonhosted.org/packages/65/47/7e02164a2a3db50ed6d8a6ab1d6d60b69c4c3fdf57a284257925dfc12bda/requests-2.19.1-py2.py3-none-any.whl
Collecting xmodem==0.4.5 (from -r labgrid/requirements.txt (line 8))
  Using cached https://files.pythonhosted.org/packages/7a/aa/3c57438250d7649af63a1ad70a790ebe353aa4c3c056e72f52a9cd75a6c9/xmodem-0.4.5-py3-none-any.whl
Collecting autobahn==19.3.3 (from -r labgrid/requirements.txt (line 9))
  Using cached https://files.pythonhosted.org/packages/9a/47/229e88fb7fc51b0ddbf7d6eca471dde206775e2f73938590fe89af011fe4/autobahn-19.3.3-py2.py3-none-any.whl
Collecting PyYAML==5.1 (from -r labgrid/requirements.txt (line 10))
  Using cached https://files.pythonhosted.org/packages/9f/2c/9417b5c774792634834e730932745bc09a7d36754ca00acf1ccd1ac2594d/PyYAML-5.1.tar.gz
Collecting ansicolors==1.1.8 (from -r labgrid/requirements.txt (line 11))
  Using cached https://files.pythonhosted.org/packages/53/18/a56e2fe47b259bb52201093a3a9d4a32014f9d85071ad07e9d60600890ca/ansicolors-1.1.8-py2.py3-none-any.whl
Collecting MarkupSafe>=0.23 (from jinja2==2.10->-r labgrid/requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/6e/57/d40124076756c19ff2269678de7ae25a14ebbb3f6314eb5ce9477f191350/MarkupSafe-1.1.1-cp35-cp35m-manylinux1_x86_64.whl
Collecting ptyprocess>=0.5 (from pexpect==4.6->-r labgrid/requirements.txt (line 3))
  Using cached https://files.pythonhosted.org/packages/d1/29/605c2cc68a9992d18dada28206eeada56ea4bd07a239669da41674648b6f/ptyprocess-0.6.0-py2.py3-none-any.whl
Collecting more-itertools>=4.0.0 (from pytest==4.0.2->-r labgrid/requirements.txt (line 5))
  Using cached https://files.pythonhosted.org/packages/b3/73/64fb5922b745fc1daee8a2880d907d2a70d9c7bb71eea86fcb9445daab5e/more_itertools-7.0.0-py3-none-any.whl
Collecting atomicwrites>=1.0 (from pytest==4.0.2->-r labgrid/requirements.txt (line 5))
  Using cached https://files.pythonhosted.org/packages/52/90/6155aa926f43f2b2a22b01be7241be3bfd1ceaf7d0b3267213e8127d41f4/atomicwrites-1.3.0-py2.py3-none-any.whl
Collecting six>=1.10.0 (from pytest==4.0.2->-r labgrid/requirements.txt (line 5))
  Using cached https://files.pythonhosted.org/packages/73/fb/00a976f728d0d1fecfe898238ce23f502a721c0ac0ecfedb80e0d88c64e9/six-1.12.0-py2.py3-none-any.whl
Collecting py>=1.5.0 (from pytest==4.0.2->-r labgrid/requirements.txt (line 5))
  Using cached https://files.pythonhosted.org/packages/76/bc/394ad449851729244a97857ee14d7cba61ddb268dce3db538ba2f2ba1f0f/py-1.8.0-py2.py3-none-any.whl
Collecting pluggy>=0.7 (from pytest==4.0.2->-r labgrid/requirements.txt (line 5))
  Using cached https://files.pythonhosted.org/packages/cc/e4/2003a5e4e445424602c8f06a436eeed9e78bcae4947128f64741c66c7216/pluggy-0.11.0-py2.py3-none-any.whl
Collecting pathlib2>=2.2.0; python_version < "3.6" (from pytest==4.0.2->-r labgrid/requirements.txt (line 5))
  Using cached https://files.pythonhosted.org/packages/2a/46/c696dcf1c7aad917b39b875acdc5451975e3a9b4890dca8329983201c97a/pathlib2-2.3.3-py2.py3-none-any.whl
Requirement already satisfied: setuptools in ./venv/lib/python3.5/site-packages (from pytest==4.0.2->-r labgrid/requirements.txt (line 5)) (20.7.0)
Collecting certifi>=2017.4.17 (from requests==2.19.1->-r labgrid/requirements.txt (line 7))
  Using cached https://files.pythonhosted.org/packages/60/75/f692a584e85b7eaba0e03827b3d51f45f571c2e793dd731e598828d380aa/certifi-2019.3.9-py2.py3-none-any.whl
Collecting chardet<3.1.0,>=3.0.2 (from requests==2.19.1->-r labgrid/requirements.txt (line 7))
  Using cached https://files.pythonhosted.org/packages/bc/a9/01ffebfb562e4274b6487b4bb1ddec7ca55ec7510b22e4c51f14098443b8/chardet-3.0.4-py2.py3-none-any.whl
Collecting idna<2.8,>=2.5 (from requests==2.19.1->-r labgrid/requirements.txt (line 7))
  Using cached https://files.pythonhosted.org/packages/4b/2a/0276479a4b3caeb8a8c1af2f8e4355746a97fab05a372e4a2c6a6b876165/idna-2.7-py2.py3-none-any.whl
Collecting urllib3<1.24,>=1.21.1 (from requests==2.19.1->-r labgrid/requirements.txt (line 7))
  Using cached https://files.pythonhosted.org/packages/bd/c9/6fdd990019071a4a32a5e7cb78a1d92c53851ef4f56f62a3486e6a7d8ffb/urllib3-1.23-py2.py3-none-any.whl
Collecting txaio>=18.8.1 (from autobahn==19.3.3->-r labgrid/requirements.txt (line 9))
  Using cached https://files.pythonhosted.org/packages/e9/6d/e1a6f7835cde86728e5bb1f577be9b2d7d273fdb33c286e70b087d418ded/txaio-18.8.1-py2.py3-none-any.whl
Installing collected packages: attrs, MarkupSafe, jinja2, ptyprocess, pexpect, pyserial, more-itertools, atomicwrites, six, py, pluggy, pathlib2, pytest, pyudev, certifi, chardet, idna, urllib3, requests, xmodem, txaio, autobahn, PyYAML, ansicolors
  Running setup.py install for pyserial: started
    Running setup.py install for pyserial: finished with status 'done'
  Running setup.py install for pyudev: started
    Running setup.py install for pyudev: finished with status 'done'
  Running setup.py install for PyYAML: started
    Running setup.py install for PyYAML: finished with status 'done'
[... 35 lines cut ...]
Collecting soupsieve>=1.2 (from beautifulsoup4->bs4)
  Using cached https://files.pythonhosted.org/packages/b9/a5/7ea40d0f8676bde6e464a6435a48bc5db09b1a8f4f06d41dd997b8f3c616/soupsieve-1.9.1-py2.py3-none-any.whl
Installing collected packages: soupsieve, beautifulsoup4, bs4
  Running setup.py install for bs4: started
    Running setup.py install for bs4: finished with status 'done'
Successfully installed beautifulsoup4-4.7.1 bs4-0.0.1 soupsieve-1.9.1
/usr/lib/python3.5/distutils/dist.py:261: UserWarning: Unknown distribution option: 'use_scm_version'
  warnings.warn(msg)
warning: install_lib: 'build/lib' does not exist -- no Python modules to install

zip_safe flag not set; analyzing archive contents...

Installed /home/labgrid-tests/labgrid/.eggs/UNKNOWN-0.0.0-py3.5.egg
Traceback (most recent call last):
  File "setup.py", line 72, in <module>
    "Programming Language :: Python :: 3.7",
  File "/usr/lib/python3.5/distutils/core.py", line 108, in setup
    _setup_distribution = dist = klass(attrs)
  File "/home/labgrid-tests/venv/lib/python3.5/site-packages/setuptools/dist.py", line 269, in __init__
    self.fetch_build_eggs(attrs['setup_requires'])
  File "/home/labgrid-tests/venv/lib/python3.5/site-packages/setuptools/dist.py", line 313, in fetch_build_eggs
    replace_conflicting=True,
  File "/home/labgrid-tests/venv/lib/python3.5/site-packages/pkg_resources/__init__.py", line 829, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'pytest-runner' distribution was not found and is required by the application

Emantor, 2019-05-24:
This looks more like an infrastructure problem or with a problem with your installed version of pypi. I'm afraid I can't help your much there. The version required by the file is definetly available on pypi.

spark5, 2019-05-24:
I always do:
pyvenv venv
source venv/bin/activate
pip install --upgrade pip
pip install -r labgrid/requirements.txt
pip install pytest-dependency
pip install pytest-html
pip install bs4 
cd labgrid
python setup.py install

Should I remove the pip install --upgrade pip ?

spark5, 2019-05-24:
When I removed pip install --upgrade pip, with my version of pip 8.1.1 labgrid failed to install.

So I added back: pip install --upgrade pip.
Then, labgrid was installed, but I always get:

/usr/lib/python3.5/distutils/dist.py:261: UserWarning: Unknown distribution option: 'python_requires'
 warnings.warn(msg)
zip_safe flag not set; analyzing archive contents...
labgrid.util.__pycache__.agentwrapper.cpython-35: module references __file__
labgrid.pytestplugin.__pycache__.hooks.cpython-35: module references __file__

Emantor, 2019-05-24:
Can you install a current version of `setuptools` into your venv?

spark5, 2019-05-24:
(venv) $ pip install setuptools
Requirement already satisfied: setuptools in ./venv/lib/python3.5/site-packages (20.7.0)

Emantor, 2019-05-24:
Please, try `pip install --upgrade setuptools`.

Emantor, 2019-06-06:
Has this been solved?

spark5, 2019-06-06:
The warning "/usr/lib/python3.5/distutils/dist.py:261: UserWarning: Unknown distribution option: 'python_requires' warnings.warn(msg)" has been solved by upgrading setuptools before running labgrid setup.py

However, I've made my script to retry 3 times the labgrid installation if something is missing.

closed 2019-07-05 by Emantor