labgrid-project/labgrid #443: Unable to get expected report as mentioned in "Strategy Fixture Example" while using uboot strategy

bhargavthriler, 2019-06-04:
Hi, I am new to labgrid, while following the example code mentioned in "Strategy Fixture Example" and replacing the barebox strategy with uboot strategy I'm get the following error
`@pytest.fixture(scope='function')     
    def shell_command(target, strategy, capsys):
        with capsys.disabled():           
>           strategy.transition('shell')  
`
Later I even tried executing the barebox example, still I faced the same issue.
`target = Target(name='main', env=Environment(config_file='strategy-example.yaml'))
strategy = BareboxStrategy(target=Target(name='main', env=Environment(config_file='strategy-example.yaml')), name=None, state=<BindingState.bound: 1>, status=<Status.unknown: 0>)
capsys = <_pytest.capture.CaptureFixture object at 0x7f1033fed0f0>
                                          
    @pytest.fixture(scope='function')     
    def shell_command(target, strategy, capsys):
        with capsys.disabled():           
>           strategy.transition('shell')  
                                          
conftest.py:27:                                     `
Am I doing some mistake, if so please explain.

My current setup is ubuntu 16.04, target platform "imx6"

Note: I didn't find any forums to discuss some issues regarding labgrid. If there is one please mention it. Sorry if this not the right place to discuss these issues.

Emantor, 2019-06-04:
Hi, your error reports contain the line where the error was logged, however they do not contain the error message. Please post the complete stacktrace.

bhargavthriler, 2019-06-05:
> Hi, your error reports contain the line where the error was logged, however they do not contain the error message. Please post the complete stacktrace.

Here's the complete stacktrace. My target is connected to "ttyUSB0".
```
Labgrid/labgrid-venv/boot-reboot2$ pytest --lg-env local.yaml test_boot.py 
=========================================================================================== test session starts ============================================================================================
platform linux -- Python 3.5.2, pytest-4.0.2, py-1.8.0, pluggy-0.12.0
rootdir: /home/bhargav/Labgrid/labgrid-venv/boot-reboot2, inifile:
plugins: labgrid-0.2.1.dev138+g9091a77, html-1.20.0, metadata-1.8.0
collected 3 items                                                                                                                                                                                          

test_boot.py main: Turn the target example-board OFF and press enterEmain: Turn the target example-board OFF and press enterEmain: Turn the target example-board OFF and press enterE                                                                                                                                                                                     [100%]

================================================================================================== ERRORS ==================================================================================================
___________________________________________________________________________________ ERROR at setup of test_uboot_initial ___________________________________________________________________________________

target = Target(name='main', env=Environment(config_file='local.yaml'))
strategy = UBootStrategy(target=Target(name='main', env=Environment(config_file='local.yaml')), name=None, state=<BindingState.bound: 1>, status=<Status.unknown: 0>)
capsys = <_pytest.capture.CaptureFixture object at 0x7f62fb7306d8>

    @pytest.fixture(scope='function')
    def bootloader_command(target, strategy, capsys):
        with capsys.disabled():
>           strategy.transition('uboot')

conftest.py:21: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../lib/python3.5/site-packages/labgrid-0.2.1.dev138+g9091a77-py3.5.egg/labgrid/strategy/ubootstrategy.py:46: in transition
    self.transition(Status.off)  # pylint: disable=missing-kwoa
../lib/python3.5/site-packages/labgrid-0.2.1.dev138+g9091a77-py3.5.egg/labgrid/strategy/ubootstrategy.py:44: in transition
    self.power.off()
../lib/python3.5/site-packages/labgrid-0.2.1.dev138+g9091a77-py3.5.egg/labgrid/binding.py:100: in wrapper
    return func(self, *_args, **_kwargs)
../lib/python3.5/site-packages/labgrid-0.2.1.dev138+g9091a77-py3.5.egg/labgrid/step.py:206: in wrapper
    _result = func(*_args, **_kwargs)
../lib/python3.5/site-packages/labgrid-0.2.1.dev138+g9091a77-py3.5.egg/labgrid/driver/powerdriver.py:52: in off
    format(name=self.name)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = Target(name='main', env=Environment(config_file='local.yaml')), msg = 'Turn the target example-board OFF and press enter'

    def interact(self, msg):
        if self.env:
>           self.env.interact("{}: {}".format(self.name, msg))
E           OSError: reading from stdin while output is captured

../lib/python3.5/site-packages/labgrid-0.2.1.dev138+g9091a77-py3.5.egg/labgrid/target.py:35: OSError
_______________________________________________________________________________________ ERROR at setup of test_shell _______________________________________________________________________________________

target = Target(name='main', env=Environment(config_file='local.yaml'))
strategy = UBootStrategy(target=Target(name='main', env=Environment(config_file='local.yaml')), name=None, state=<BindingState.bound: 1>, status=<Status.unknown: 0>)
capsys = <_pytest.capture.CaptureFixture object at 0x7f62fb68a5f8>

    @pytest.fixture(scope='function')
    def shell_command(target, strategy, capsys):
        with capsys.disabled():
>           strategy.transition('shell')

conftest.py:27: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../lib/python3.5/site-packages/labgrid-0.2.1.dev138+g9091a77-py3.5.egg/labgrid/strategy/ubootstrategy.py:53: in transition
    self.transition(Status.uboot)
../lib/python3.5/site-packages/labgrid-0.2.1.dev138+g9091a77-py3.5.egg/labgrid/strategy/ubootstrategy.py:46: in transition
    self.transition(Status.off)  # pylint: disable=missing-kwoa
../lib/python3.5/site-packages/labgrid-0.2.1.dev138+g9091a77-py3.5.egg/labgrid/strategy/ubootstrategy.py:44: in transition
    self.power.off()
../lib/python3.5/site-packages/labgrid-0.2.1.dev138+g9091a77-py3.5.egg/labgrid/binding.py:100: in wrapper
    return func(self, *_args, **_kwargs)
../lib/python3.5/site-packages/labgrid-0.2.1.dev138+g9091a77-py3.5.egg/labgrid/step.py:206: in wrapper
    _result = func(*_args, **_kwargs)
[... 23 lines cut ...]
conftest.py:21: 
[... cut ...]
    def interact(self, msg):
        if self.env:
>           self.env.interact("{}: {}".format(self.name, msg))
E           OSError: reading from stdin while output is captured

../lib/python3.5/site-packages/labgrid-0.2.1.dev138+g9091a77-py3.5.egg/labgrid/target.py:35: OSError
========================================================================================= 3 error in 0.16 seconds ==========================================================================================
(labgrid-venv)
```

Emantor, 2019-06-05:
Pytest requires the `-s` option if the `ManualPowerDriver` is used. Please try again with the `-s` parameter.

bhargavthriler, 2019-06-06:
> Pytest requires the `-s` option if the `ManualPowerDriver` is used. Please try again with the `-s` parameter.

Thank You.
Yes it works now. 
Seems like Readme needs to be updated.

Emantor, 2019-06-06:
Yes, can you create a pull request?

bhargavthriler, 2019-06-06:
> Yes, can you create a pull request?

Created

bhargavthriler, 2019-06-20:
@Emantor is there tutorial or guide to setup imx6 boards in labgrid and get the uart logs as well, because the documentation doesn't has much information for getting the uart logs. If there is IRC channel or mailing list that would also help. As I am stuck while making a soft reboot strategy. 
Thank you!

bhargavthriler, 2019-06-26:
> @Emantor is there tutorial or guide to setup imx6 boards in labgrid and get the uart logs as well, because the documentation doesn't has much information for getting the uart logs or creating our own strategies. If there is IRC channel or mailing list that would also help. As I am stuck while making a soft reboot strategy.
> Thank you!

@Emantor I needed  some help?

Emantor, 2019-06-26:
You probably want `--lg-log`. As for creating your own strategies, I'd start by copying one of the example strategies and changing that. You can import other python files using the `imports:` list in the YAML file.
i.e.
```
targets:
  main:
    resources:
      RawSerialPort:
        port: "/dev/ttyUSB0"
    drivers:
      SerialDriver: {}
      ShellDriver:
        prompt: 'root@\w+:[^ ]+ '
        login_prompt: ' login: '
        username: 'root'
      MyFancyStrategy: {}
imports:
  - myfancystrategy.py
```

Emantor, 2019-07-05:
This is now tracked in https://github.com/labgrid-project/labgrid/issues/443

closed 2019-07-05 by Emantor