labgrid-project/labgrid #457: how to access uboot prompt or shell after a reboot by 'reboot' command or any other reboot scenarios from the shell

bhargavthriler, 2019-06-28:
I'm trying to reboot the target using 'reboot' command (form shell) or 'reset' command (from uboot) within a test. But after reboot or the target directly boot from it's default boot methods and reaches login prompt but can't login and looses it's track of remaining tests. 

Check the below test scenerio 

```
def test_shell(shell_command):
    stdout = shell_command.run_check('ls -la /')

def test_shell(shell_command):
    stdout = shell_command.run_check('reboot -h now')

def test_shell(shell_command):
    stdout = shell_command.run_check('uname -a')

```
or 

```
def test_uboot_initial6(bootloader_command):
     bootloader_command.run_check('version')

def test_uboot_after_reboot2(bootloader_command):
    bootloader_command.run_check('reset')

def test_uboot_after_reboot5(bootloader_command):
    bootloader_command.run_check('help')

```
In above test scenarios  after executing the second test the target loses track of it's state and can't execute and last test.

Is there any way to make this type of test possible ?

Bastian-Krause, 2019-06-28:
Sure, those types of tests are possible. What does your `shell_command`/`bootloader_command` fixture look like?

bhargavthriler, 2019-06-28:
> Sure, those types of tests are possible. What does your `shell_command`/`bootloader_command` fixture look like?
> 
> Edit: please note that `run()` or `run_check()` might not work in these cases. At least for the bootloader reset case there will be no new prompt after `reset` thus no way to check the return code of the command.

Here is my 'conftest.py' file
```
import pytest

from labgrid.protocol import CommandProtocol
#from labgrid.strategy import UBootRebootStrategy
import UBootRebootStrategy
from labgrid.exceptions import NoDriverFoundError

@pytest.fixture(scope='session')
def strategy(target):
    try:
        return target.get_driver('UBootRebootStrategy')
    except NoDriverFoundError:
        pytest.skip("strategy not found")

@pytest.fixture(scope='function')
def switch_off(target, strategy, capsys):
    with capsys.disabled():
        strategy.transition('off')

@pytest.fixture(scope='function')
def bootloader_command(target, strategy, capsys):
    with capsys.disabled():
        strategy.transition('uboot')
    return target.get_active_driver(CommandProtocol)

@pytest.fixture(scope='function')
def shell_command(target, strategy, capsys):
    with capsys.disabled():
        strategy.transition('shell')
    return target.get_active_driver(CommandProtocol)

```
Can you tell me how to check for the login prompt and pass a predefined set of commands when the target reboots for continuing the remaining tests.?

bhargavthriler, 2019-07-10:
> Sure, those types of tests are possible. What does your `shell_command`/`bootloader_command` fixture look like?
>
> Edit: please note that `run()` or `run_check()` might not work in these cases. At [...]

@Bastian-Krause Check my 'bootloader_command' and 'shell_command' fixtures

bhargavthriler, 2019-07-22:
@Emantor Can you help me with this?

alegsan, 2021-01-12:
@bhargavthriler could you solve the problem meanwhile? currently i'am in the same situation. I want to write a test where i need to exec the `reboot` command and wait for the login-prompt. I need to use the `reboot` command explicitly because only in this case we get a specific failure in some situations. But `run()`/`run_check()` get a timeout because the prompt does not appear anymore.

@Emantor or could you give us a small example how to solve this problem?

Bastian-Krause, 2021-01-13:
Sorry for the delay. In a strategy this could look like this:
```
self.console.sendline("reset")
self.target.deactivate(self.uboot)
self.target.activate(self.uboot)
```
`self.uboot` is an `UBootDriver` instance, `self.console` is a `ConsoleProtocol` instance. [sendline()](https://github.com/labgrid-project/labgrid/blob/c8756f22952ce8bed7bb621185afd013e0b3273a/labgrid/driver/consoleexpectmixin.py#L44) uses pexpect's spawn [sendline()](https://pexpect.readthedocs.io/en/stable/api/pexpect.html#pexpect.spawn.sendline). This way you can send a command without waiting for it to complete in the U-Boot shell. Reactivating the UBootDriver makes sure U-Boot comes up again.

`BareboxDriver.reset()` does [something similar](https://github.com/labgrid-project/labgrid/blob/b40c7380a2da8a278b1881a072fbaa090a1639ef/labgrid/driver/bareboxdriver.py#L99).

alegsan, 2021-01-13:
@Bastian-Krause thanks for the quick response....

i will try your suggestion today...

i mixed some things up in my response...but i suppose your suggestion also fits for the shell driver (when i want to exec the `reboot` command on the linux shell) in combination with the serial driver. Am i correct?

Emantor, 2021-01-13:
> @Bastian-Krause thanks for the quick response....
>
> i will try your suggestion today...
>
> i mixed some things up in my response...but i suppose your suggestion also fits for the shell driver (when i [...]

Yes. Thats why we recommend using a strategy, which will simply traverse the same steps as on the first boot after the reset.

alegsan, 2021-01-13:
> Yes. Thats why we recommend using a strategy, which will simply traverse the same steps as on the first boot after the reset.

great..thanks for your help

Bastian-Krause, 2021-01-22:
Glad I could help. Closing this, if something is not clear yet, please reopen.

closed 2021-01-22 by Bastian-Krause