labgrid-project/labgrid #464: NetworkPowerPort should accept user and password for PDUs

spark5, 2019-07-05:
In case of a PDU = Power Distribution Unit, like EG-PMS2-LAN or DLI Web Power Switch, a custom user and password can be configured.

Emantor, 2019-07-05:
Depending on the security level required, a new power backend can be added and the parameters could be supplied by the YAML environment file.

jluebbe, 2019-07-08:
I'd prefer to handle new power controllers via https://github.com/pdudaemon/pdudaemon

spark5, 2019-07-08:
Ok. Then, are you willing to add another power driver in labgrid?

jluebbe, 2019-07-08:
We'd take a PR to add pdudaemon support. It's been on my list for some time, but more urgent stuff keeps coming up. If you're interested, I could sketch the resource and driver structure.

spark5, 2019-07-08:
Yes, I'm interested.

sephalon, 2019-11-19:
Has anybody started with PDUDaemon support yet? We also need support for two different new PDUs and I probably could spend some time working on this.

Emantor, 2019-11-19:
See #510, pdudaemon should already be working.

sephalon, 2019-11-19:
Great, somehow I missed that…

Emantor, 2024-03-20:
PDUDaemon is the recommended way here.

closed 2024-03-20 by Emantor