labgrid-project/labgrid #489: labgrid-client commands fail but coordinator running

spark5, 2019-08-30:
When running labgrid-client commands, the coordinator is up and running, exporter the same, but this is thrown:

```
 ERROR: Task exception was never retrieved
future: <Task finished coro=<onJoin() done, defined at labgrid-tests/venv/lib/python3.5/site-packages/labgrid-0.1.dev1453+g1b56ade.d20190830-py3.5.egg/labgrid/remote/client.py:69> exception=ApplicationError('no callee registered for procedure <org.labgrid.coordinator.get_resources>',)>
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/tasks.py", line 241, in _step
    result = coro.throw(exc)
  File "labgrid-tests/venv/lib/python3.5/site-packages/labgrid-0.1.dev1453+g1b56ade.d20190830-py3.5.egg/labgrid/remote/client.py", line 72, in onJoin
    'org.labgrid.coordinator.get_resources'
  File "/usr/lib/python3.5/asyncio/futures.py", line 361, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.5/asyncio/tasks.py", line 296, in _wakeup
    future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 274, in result
    raise self._exception
autobahn.wamp.exception.ApplicationError: ApplicationError(error=<wamp.error.no_such_procedure>, args=['no callee registered for procedure <org.labgrid.coordinator.get_resources>'], kwargs={}, enc_algo=None, callee=None, callee_authid=None, callee_authrole=None, forward_for=None)
```

The partial output from coordinator:

```
aug 30 14:01:03 : 2019-08-30T14:00:57+0300 [Router      19901 crossbar.router.protocol.WampWebSocketServerProtocol] Connection to/from tcp4:IP:45296 was closed cleanly
aug 30 14:01:03 : 2019-08-30T14:00:57+0300 [Router      19901 crossbar.router.protocol.WampWebSocketServerProtocol] _connectionLost: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionDone'>: Connection was closed cleanly.
aug 30 14:01:03 : ]
aug 30 14:01:03 : 2019-08-30T14:00:57+0300 [Router      19901 crossbar.router.protocol.WampWebSocketServerProtocol] WAMP-over-WebSocket transport lost: wasClean=True, code=1000, reason="None"
```

jluebbe, 2019-08-30:
I suspect that the coordinator component in the crossbar process didn't start correctly. Do you see an exception while starting crossbar?

spark5, 2019-08-30:
After a while, the above coordinator error is thrown. I've restarted coordinator 20 minutes ago.
I also have debug mode enabled in coordinator.

jluebbe, 2019-08-30:
The exporter output has truncated lines and the coordinator output is not conclusive. Could you put the full crossbar/coordinator output in a github gist?

spark5, 2019-08-30:
There is a lot of output because i have debug mode enabled. Maybe I can give more output before and after the error?

jluebbe, 2019-08-30:
> There is a lot of output because i have debug mode enabled. Maybe I can give more output before and after the error?

The large amount of output was the reason I suggested using a gist (https://gist.github.com/). Please provide the full output from crossbar start until ``Coordinator ready.``.

spark5, 2019-08-30:
The output is here: https://gist.github.com/spark5/0d765f1dba222c6624ebd4f154a9b3ba

It contains everything until Coordinator ready.
I didn't see any error or exception in that output.

However, the error i've pasted above this post is after Coordinator ready. 
Here you have the error output just before next same error:
https://gist.github.com/spark5/b7f6f6e4fcc5be4a9120167c6bd09260

jluebbe, 2019-08-30:
``autobahn.wamp.exception.ApplicationError: ApplicationError(error=<wamp.error.no_such_procedure>, args=['no callee registered for procedure <org.labgrid.coordinator.get_resources>'], kwargs={}, enc_algo=None, callee=None, callee_authid=None, callee_authrole=None, forward_for=None)
`` indicates that the coordinator is not running or that it has not registered correctly. However, in you second log, when you search for "Guest" (which is the output from the coordinator), you see that it is running and processing requests like ``org.labgrid.coordinator.set_resource``. I don't see any reason why get_resources should behave differently.
Could you check that you are on current master and the dependencies are as specified in the requirements.txts?
You could also add some instrumentation to the ``coordinator.py`` find out where it fails.

spark5, 2019-08-30:
I don't have latest labgrid code. My requirements.txt has:
```
attrs==19.1.0
jinja2==2.10.1
pexpect==4.7
https://github.com/labgrid-project/pyserial/archive/v3.4.0.1.zip#egg=pyserial
pytest==4.6.3
pyudev==0.21.0
requests==2.22.0
xmodem==0.4.5
autobahn==19.3.3
PyYAML==5.1.1
ansicolors==1.1.8
pyusb==1.0.2
```

jluebbe, 2019-09-03:
Please use the latest code and apply crossbar-requirements.txt as well.

spark5, 2019-09-04:
Hi!

With latest code: remote/client: fix strategy support with console command,
the error is thrown again. The coordinator basically stays in a loop with that error. This situation occurred since we've added 20+ boards.

Can you please investigate this issue?

jluebbe, 2019-09-04:
From you logs I don't see any reason why it would behave that way, so my options to investigate are very limited. We don't see issues like this in our lab with ~200 places.
Do you see that exception on all exporters? Did you try to instrument the coordinator startup around the lines where it is registering it's API functions?

spark5, 2019-09-05:
Hi!

The way I am using all 20+ places simultaneously is by checking if some board places of some type are available. So I basically check in alphabetically order each one. I don't think I've reached the maxim number of connections the crossbar.io has.
Once the coordinator is in that error loop, all the exporters have the same issue: no RPC call exists.
I have debug mode enabled in coordinator. Any ideas?

spark5, 2019-09-26:
Hi!
Now I am using labgrid-client places -a and I acquire a board only when I know it is free.
Before, the crossbar server was being spammed with a lot of connections with retries after 20 seconds.

The coordinator is up and running now and no other crashes occurred.

Emantor, 2020-01-10:
Your last message indicates that this is no longer an issue, closing. Please reopen if this is still a problem.

closed 2020-01-10 by Emantor

jluebbe, 2020-01-10:
You could also try using the reservation mechanism (https://github.com/labgrid-project/labgrid/pull/483) with tags for the board type.

atline, 2022-01-26:
> Before, the crossbar server was being spammed with a lot of connections with retries after 20 seconds.

@spark5 @jluebbe @Emantor  , sorry, but I encountered similar problems here. 

Everything is ok when crossbar start, but after runs for some time. If I use next command, all reports error:

**1. labgrid-client who**

```
$ labgrid-client who
  ERROR: Task exception was never retrieved
future: <Task finished coro=<ClientSession.onJoin() done, defined at /home/nxa13855/labgrid-venv/lib/python3.7/site-packages/labgrid/remote/client.py:76> exception=ApplicationError('no callee registered for procedure <org.labgrid.coordinator.get_resources>')>
Traceback (most recent call last):
  File "/home/nxa13855/labgrid-venv/lib/python3.7/site-packages/labgrid/remote/client.py", line 79, in onJoin
    'org.labgrid.coordinator.get_resources'
autobahn.wamp.exception.ApplicationError: ApplicationError(error=<wamp.error.no_such_procedure>, args=['no callee registered for procedure <org.labgrid.coordinator.get_resources>'], kwargs={}, enc_algo=None, callee=None, callee_authid=None, callee_authrole=None, forward_for=None)
```

**2. labgrid-client places**

```
  ERROR: Task exception was never retrieved
future: <Task finished coro=<ClientSession.onJoin() done, defined at /home/nxa13855/labgrid-venv/lib/python3.7/site-packages/labgrid/remote/client.py:76> exception=ApplicationError('no callee registered for procedure <org.labgrid.coordinator.get_resources>')>
Traceback (most recent call last):
  File "/home/nxa13855/labgrid-venv/lib/python3.7/site-packages/labgrid/remote/client.py", line 79, in onJoin
    'org.labgrid.coordinator.get_resources'
autobahn.wamp.exception.ApplicationError: ApplicationError(error=<wamp.error.no_such_procedure>, args=['no callee registered for procedure <org.labgrid.coordinator.get_resources>'], kwargs={}, enc_algo=None, callee=None, callee_authid=None, callee_authrole=None, forward_for=None)
```

**3. All exporter linked labgrid coordinator reports:**

```
{'cls': 'NetworkSerialPort', 'params': OrderedDict([('host', '10.192.244.147'), ('port', 8602), ('extra', {'proxy': 'LAVA-Debug', 'proxy_required': False})]), 'acquired': None, 'avail': True}
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/labgrid-0.4.1.dev46+ga19d78d-py3.7.egg/labgrid/remote/exporter.py", line 719, in onJoin
    group_name, resource_name, cls, params
  File "/usr/local/lib/python3.7/dist-packages/labgrid-0.4.1.dev46+ga19d78d-py3.7.egg/labgrid/remote/exporter.py", line 819, in add_resource
    await self.update_resource(group_name, resource_name)
  File "/usr/local/lib/python3.7/dist-packages/labgrid-0.4.1.dev46+ga19d78d-py3.7.egg/labgrid/remote/exporter.py", line 828, in update_resource
    data
autobahn.wamp.exception.ApplicationError: ApplicationError(error=<wamp.error.no_such_procedure>, args=['no callee registered for procedure <org.labgrid.coordinator.set_resource>'], kwargs={}, enc_algo=None, callee=None, callee_authid=None, callee_authrole=None, forward_for=None)
connection lost
```

After I restart `labgrid coordinator` everything ok.

NOTE: I deploy both coordinator and exporter in docker, when exporter exits I will use `--restart=always` to try to reconnect to `labgrid coordinator`. Isn't that may cause more connections to `coordinator` which similar to what @spark5 mentioned?

Anyway, this error looks more related to `autobahn.wamp` which is also a thirdparty communication firmware not develop by you? Then, do you know at what kinds of situation, such `no callee registered for procedure` will be shown?

Please help, thanks.

Emantor, 2022-01-27:
Labgrid internally uses the crossbar WAMP infrastructure for the remote procedure calls. When starting the coordinator, two processes are started:

1. the WAMP crossbar router which handles connections from clients and forwards the RPC calls
2. the labgrid coordinator component which handles calls from clients

The traces you provide above indicate that the WAMP crossbar router is still running, however the labgrid coordinator component is no longer available. We have seen similar behaviour in our own deployment on a virtual machine where the underlying IO layer blocks for several seconds, the labgrid coordinator uses a websocket connection to connect to the router and if pings from the router are not answered in time the router simply disconnects the coordinator. This should also be seen within the crossbar logging output, something like:
```
2020-01-06 20:41:58 labgrid crossbar[21594]: [Guest       21639 crossbar.node.worker.GuestWorkerProcess] 2020-01-06T20:41:58 dropping connection to peer tcp6:::1:20408 with abort=True: WebSocket ping timeout (peer did not respond with pong in time)
```

indicates that the router disconnected the coordinator. We fixed our underlying IO blocking problems and this seemingly fixed the problems.