labgrid-project/labgrid #492: create a labgrid-client to list the available exporters

spark5, 2019-09-11:
Hi!

I would be helpful  when checking exporters healthiness to have a labgrid-client command that is able to list the available exporters. 
What do you think?

Emantor, 2019-09-20:
This kind of already exists in `labgrid-client resources` where each resource is prefixed with the exporter name. `labgrid-client resources | awk -F "/" '{ print $1 }' | uniq` should print all available exporters.

spark5, 2019-09-20:
What I imagined this command to look like was to list the available exporters like status = up/down
I know that crossbar has internal heartbeats so you can rely on the fact that once an exporter is down, you save the exporter status in coordinator's resources.yaml.

What do you think?

Emantor, 2019-09-20:
Currently the coordinator does not track the exporter information, this would mean that any exporter which was once connected to the coordinator will show up as unavailable. Not a problem in fixed setups, but might be unintuitive in setups where developer computers may sometimes be in the system and other times not.

spark5, 2019-09-20:
Alright! I've missed the case of developer exporter not being available all the time.
So I guess just a snapshot of the current exporters will be enough.

jluebbe, 2020-01-10:
Yes, this sounds useful. We should also keep track of some statistics in the coordinator and report those as well (such as: uptime, event count, number of resources).