labgrid-project/labgrid #493: all places corresponding to an exporter are visible but no resources are exported

spark5, 2019-09-16:
Hi!

I happened that the exporter crashed but it was restarted as a systemctl service with a configuration having:
```
Restart=always
RestartTimeout=5
```
The exporter was up and running but not exporting any resources in labgrid-client resources command.
However, the coordinator was able to see the places corresponding to that exporter.

Can you please investigate? Is it a normal flow?

Thanks!

Emantor, 2019-09-16:
We have no way to investigate this issue since we have neither the logs from the coordinator nor from the exporter.

spark5, 2019-09-17:
Hi!

I have changed the way systemctl service is restarting: ```Restart=on-failure```
Now, when exporter skips a heartbeat or there is a timeout in async operation, the systemctl service is no longer restarting. So the exporter is up and running all the time.

According to doc: [https://www.freedesktop.org/software/systemd/man/systemd.service.html#Restart=](url) 'always' is a superset of 'on-failure'.

closed 2019-09-17 by spark5