labgrid-project/labgrid #497: Error SSH connection with NetworkService

qgrembert, 2019-09-21:
HI!

I am starting to use labgrid.
As a first test, I tried to execute test over ssh on my computer (localhost).
I used the following env file : 

```
targets:
  main:
    resources:
      NetworkService:
        address: 'localhost'
        username: 'quentin'
    drivers:
      ShellDriver:
        prompt: 'quentin@debian:~ '
        login_prompt: 'debian login: '
        username: 'quentin'
      SSHDriver: {}
```
conftest.py : 
```
@pytest.fixture(scope='session')
def command(target):
    ssh = target.get_driver('SSHDriver')
    target.activate(ssh)
    shell = target.get_driver('ShellDriver')
    target.activate(shell)
    return shell
```

When running a test with` pytest --lg-env local.yaml test_shell.py`, I have the following error :: 
`labgrid.exceptions.NoDriverFoundError: no driver matching <class 'labgrid.protocol.consoleprotocol.ConsoleProtocol'> found in target Target(name='main', env=Environment(config_file='local.yaml'))`

If I add a SerialDriver (like in the shell example), the problem disappear .

The problem do not come from the SSH authentication : the correct key is set in .ssh/authorized_keys. Il I remove the key from this file, I have the following error : 
labgrid.driver.exception.ExecutionError: failed to connect to localhost with ['sshpass', '-e', 'ssh', '-f', '-o', 'LogLevel=ERROR', '-x', '-o', 'ConnectTimeout=30', '-o', 'ControlPersist=300', '-o', 'UserKnownHostsFile=/dev/null', '-o', 'StrictHostKeyChecking=no', '-o', 'ServerAliveInterval=15', '-MN', '-S', '/tmp/labgrid-ssh-tmp-zzbb4daq/control-localhost', '-p', '22', 'quentin@localhost'] and 5

I don't know if it is a bug or I'm not doing things correctly.

All help is welcome.
Thank's.

Emantor, 2019-09-22:
The `SSHDriver` implements the `CommandProtocol`, which is the same protocol as implemented by the `ShellDriver`. The `ShellDriver however requires a `ConsoleProtocol`, since it converts a text based serial interface into a command execution interface. If you only have SSH, remove the `Shelldriver`

Emantor, 2019-12-06:
Closing this, if this is still an issue please comment.

closed 2019-12-06 by Emantor