labgrid-project/labgrid #499: exporters being kicked out after polling

spark5, 2019-09-26:
Hi!

When coordinator is polling the exporters, after some time, some exporters are being restarted because they loose connection.
In coordinator, there are multiple pings:
'''crossbar.router.protocol.WampWebSocketServerProtocol] WebSocketProtocol.onPing(payload=<4 bytes>)'''
but the exporter says "peer did not respond with pong in time".

I've tried reducing the numbers of ping/pongs within the timeout of polling by configuring ```auto_ping_interval : 5000```, but it seems that the problem persists.

Would it be a good idea to double the timeout when polling?

spark5, 2019-10-15:
Hi!

After some investigations about autobahn, I see that there are two implementations for the ApplicationRunner: autobahn.twisted.wamp and autobahn.asyncio.wamp.

You are using the asyncio version in order to modify the socket options to add TCP_NODELAY option. The twisted version doesn't allow getting the socket.

After reading some autobahn examples, it seems that autobahn.twisted has an autoreconnect parameter when starting ApplicationRunner. Wouldn't it be ok to switch from asyncio to twisted in order to make exporter autoreconnect itself when ping-pongs were missed due to timeout?

This would mean that exporter has a failure mechanism, instead of just shutting down itself immediately. Currently, I have my exporter as a systemctl service which is restarted in 5 seconds when onDisconnect() was triggered.

Can you please add some failure mechanism for exporter?

My boards are running a lot of reboot tests which make the USB OTG resources to be enumerated multiple times. So when the exporter is occupied with monitoring USBs, it is missing ping-pong communication with coordinator.

jluebbe, 2019-10-15:
As labgrid uses asyncio already, we can't use twisted for autobahn.
Reconnecting without recreating the whole state at the coordinator would mean that there is a risk of lost messages, so it can't be significantly faster than restarting the exporter.

Did you see a change in behaviour from c4613f307b1e9911523a7ee04d80627d07709b8f? This was intended for fix this issue.

spark5, 2019-10-15:
Hi!

My latest labgrid version is from 9 aug 2019. So I have that commit from the very first beginning. 
I started using labgrid since january 2019. 

As soon as I used more CI plans in parallel, I noticed that the coordinator was overwhelmed with connections from my management script. After I changed the script, the coordinator stabilized, so the exporters. I made sure that my exporters' systemctl service is restarted only on failure. So a timeout won't make the systemctl to restart an exporter.

Currently, the exporters are crashing almost each day. I am adding more and more reboot tests and CI plans. Maybe because I am stressing labgrid it is facing this kind of issues?

Did TCP_NODELAY setting make labgrid communication to be more stable? I am asking because I didn't see this into autobahn examples.

jluebbe, 2019-10-15:
> My latest labgrid version is from 9 aug 2019. So I have that commit from the very first beginning.
> I started using labgrid since january 2019.
OK, I haven't seen disconnects like you describe since that fix.
 
> As soon as I used more CI plans in parallel, I noticed that the coordinator was overwhelmed with connections from my management script. After I changed the script, the coordinator stabilized, so the [...]
I think we currently have restart=always.

> Currently, the exporters are crashing almost each day. I am adding more and more reboot tests and CI plans. Maybe because I am stressing labgrid it is facing this kind of issues?

Hmm. How many exporters do you have? How many tests are running at the same time? What systems are you using as exporters (we have PCEngines APUs)?

> Did TCP_NODELAY setting make labgrid communication to be more stable? I am asking because I didn't see this into autobahn examples.

It reduced the startup time of exporters, because it reduces the round-trip-time of RPC calls to the coordinator.

spark5, 2019-10-16:
I've also used restart=always, but it was long time ago. After I discovered that an exporter async operation can timeout, I've modified my systemctl service to have restart=on-failure.
Restart=always makes an exporter to restart even when a signal was triggered inside exporter or when exporter exited cleanly.
This is not something I would want to have for my CI. So I want exporters to restart only on failure.

I have 4 active exporters used by CI, but in total there are 7 of them, with a total of 60+ boards. Each board can run hundreds of tests which also include reboot tests. The exporters are running as systemctl services inside commodity workstations. 

You said that switching to twisted.wamp would require also a change in coordinator behavior.
Would you consider including an autoreconnect mechanism for exporter with the currently used autobahn.asyncio.wamp ?

I found some related stuff here: [https://github.com/isra17/autobahn-autoreconnect/blob/master/autobahn_autoreconnect/__init__.py](url)

And also here: [https://github.com/crossbario/autobahn-python/issues/295](url).
Relevant comment is this one: 
> There is a solution if you use the newer component API; both asyncio and twisted versions of this do auto-reconnect (with backoff and jitter etc). See examples/{asyncio,twisted}/wamp/component/* for some examples.

jluebbe, 2020-01-10:
As automatic reconnects would still lead to missed events, that approach is unsuitable to labgrid. To keep the local state of places and resources in sync the other hosts, skipping events would be fatal.

Emantor, 2020-10-26:
I'll close this as answered, reopen if necessary.

closed 2020-10-26 by Emantor