labgrid-project/labgrid #502: Acquired places prevent exporter from reregistering resources after restart

Emantor, 2019-10-02:
1. Coordinator and Exporter are running
2. User locks a board which acquires resources from the exporter
3. Exporter requires are restart due to changed configuration
4. Shutdown of the exporter results in the following messages:
```
Oct 02 10:38:56 labgrid crossbar[16844]: [Guest       16856] failed to release ResourceImport(data={'cls': 'NetworkSerialPort', 'avail': True, 'acquired': '10015', 'params': {'port': 4015, 'host': '192.168.23.5'}}, path=('rl10-srv', '10015', 'NetworkSerialPort', 'NetworkSerialPort'))
```
5. After restart the resources from the exporter are reregistered at the coordinator but do not show up in client calls, shows of places produce:
```
> lgc -p device show                                                                                                                                                                                  ✔  10:48
Place 'device':
  aliases: some-alias
  matches:
    rl10-srv/rl10-usb-oben-p8/NetworkSerialPort
    rl10-srv/10016/NetworkPowerPort
    rl10-srv/rl10-usb-oben-p5/NetworkIMXUSBLoader
    rl10-srv/wago-slice3-do1/ModbusTCPCoil
    rl10-srv/rl10-usb-oben-p5/NetworkAndroidFastboot/AndroidFastboot
  acquired: some-computer/rcz
  acquired resources:
    rl10-srv/10016/NetworkPowerPort/NetworkPowerPort
    rl10-srv/rl10-usb-oben-p5/NetworkAndroidFastboot/AndroidFastboot
    rl10-srv/rl10-usb-oben-p5/NetworkIMXUSBLoader/IMXUSBLoader
    rl10-srv/rl10-usb-oben-p8/NetworkSerialPort/USBSerialPort
    rl10-srv/wago-slice3-do1/ModbusTCPCoil/ModbusTCPCoil
  created: 2019-01-21 12:11:21.504197
  changed: 2019-10-02 10:48:32.111486
Traceback (most recent call last):
  File "/usr/ptx-venvs/labgrid/lib/python3.7/site-packages/labgrid/remote/client.py", line 1529, in main
    session.loop.run_until_complete(coro)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "/usr/ptx-venvs/labgrid/lib/python3.7/site-packages/labgrid/remote/client.py", line 360, in print_place
    resource = self.resources[exporter][group_name][resource_name]
KeyError: 'rl10-srv'
```

We should release the acquired resources on the exporter when the exporter disconnects unconditionally. Meanwhile the workaround is to release all places with acquired resources from the exporter and restarting the exporter, than registration works correctly again.

Bastian-Krause, 2023-07-23:
I've never seen this. Does this still happen?

Emantor, 2024-03-20:
I haven't seen this issue in a while and our internal logs contain no mention of this, closing.

closed 2024-03-20 by Emantor