labgrid-project/labgrid #526: Exporter suspends lead to labgrid call hangs

Bastian-Krause, 2019-11-06:
All labgrid calls hang for 10 minutes if a system running an exporter is suspended. After that everything works again as expected.

The following logs can be observed after the system is woken up again:
```
2019-11-05T17:30:38 dropping connection to peer tcp:2001:67c:670:100:1d::22:20408 with abort=True: WebSocket ping timeout (peer did not respond with pong in time)
connection lost
2019-11-05T17:30:38 Exception in callback _AsyncioApi.create_future.<locals>.done(<Future cancelled>) at /usr/ptx-venvs/labgrid/lib/python3.5/site-packages/txaio/aio.py:407
handle: <Handle _AsyncioApi.create_future.<locals>.done(<Future cancelled>) at /usr/ptx-venvs/labgrid/lib/python3.5/site-packages/txaio/aio.py:407>
Traceback (most recent call last):
  File "/usr/ptx-venvs/labgrid/lib/python3.5/site-packages/txaio/aio.py", line 409, in done
    f.exception()
  File "/usr/lib/python3.5/asyncio/futures.py", line 305, in exception
    raise CancelledError
concurrent.futures._base.CancelledError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
    self._callback(*self._args)
  File "/usr/ptx-venvs/labgrid/lib/python3.5/site-packages/txaio/aio.py", line 411, in done
    canceller(f)
  File "/usr/ptx-venvs/labgrid/lib/python3.5/site-packages/autobahn/wamp/protocol.py", line 1611, in canceller
    self._transport.send(cancel_msg)
AttributeError: 'NoneType' object has no attribute 'send'
2019-11-05T17:30:38 session closed with reason wamp.close.transport_lost [WAMP transport was lost without closing the session before]
2019-11-05T17:30:38 Cancelling 1 outstanding requests
```
Might be related to #514.

Emantor, 2019-12-06:
@jluebbe has a patch series queued up which should fix this.

Emantor, 2020-01-10:
Should be fixed by #547

closed 2020-01-10 by Emantor